PX4/PX4-Autopilot #6227: Unreliable wifi connection on pixracer

CarlOlsson, 2017-01-01:
If I try to connect a device to a Pixracer the green bar in QGC indicating the status of loaded parameters will not reach 100% so I am not able to configure anything and the connection constantly drops and connects again.
This is with 1.5.3 release and with current master 9178bb73712bf824c41c8deebc632f409b3cfa3a

I have tried with two different Pixracers, two different wifi modules, Iphone, Ipad and Laptop, all with the same results.

However, If I flash [v1.4.4 release](https://github.com/PX4/Firmware/releases/tag/v1.4.4) it seems to work perfectly.

LorenzMeier, 2017-01-01:
You need to update your QGC version to a recent daily build. Its an incompatibility with the wireless modem.

closed 2017-01-01 by LorenzMeier

CarlOlsson, 2017-01-01:
Yes you are right, now it works on my laptop. I guess the patch will also come in an update soon to QGC for iOS.