PX4/PX4-Autopilot #6231: Oled Support via i2c

edsellar, 2017-01-02:
Adding an oled driver to put a small ~1" screen onboard would be great for seeing the current status at the vehicle. Colored leds are good but these are cheap and really easy to use. Good when not with a ground station. Someone already wrote the driver for arduoilot integration. Point me in the right direction and I'll give it a try but I'm new to this and think it would be a very good addition. 

https://github.com/ArduPilot/ardupilot/pull/5082

LorenzMeier, 2017-01-02:
Would certainly be interesting. Here is a good start for you - the driver is I2C and outputs the system status:
https://github.com/PX4/Firmware/blob/master/src/drivers/bst/bst.cpp

Note that we can only accept BSD 3-clause compatible code - so you must not copy any code from the ArduPilot driver.

edsellar, 2017-01-05:
Thanks - ok I'll give this a go and see if I can make sense of it. Noted on the not copying code.

edsellar, 2017-01-09:
@lorenzmeier I've started on this. We'll see what comes of it in terms of functionality. Looking at the init phase of this (pretty new to all this but I've pulled a few other i2c drivers and display drivers). Been trying to make some correlation between how this was done for apm and the appropriate redo for px4 as that's all I have to go on. A lot of the code comes from an ardufruit library so just trying To piece this together.

mhkabir, 2017-01-09:
What is the license on the Adafruit lib?

magicrub, 2017-01-09:
Which one?

edsellar, 2017-01-09:
https://github.com/adafruit/Adafruit_SSD1306

It's listed as BSD. Doesn't say if 3-clause. But says it's BSD.

kd0aij, 2017-01-09:
the page you linked says "BSD 3-clause" right above the green "clone" button

edsellar, 2017-01-10:
Good it's there. Shows how well I can't read then. :( Time for glasses.

scubAssist, 2017-02-27:
Hello, before i put any time in this myself, i wanted to ask if this has been done yet?
I'm new to the PX4 arena, but i have a spare SSD1306 lying around which i wanted to hook up. The leds and beeps are all very nice, but i would like some more detailed messages without the need for a GCS.
If it has **not** been done, please let me know anyway!

edsellar, 2017-03-11:
@scubAssist looked at it for a bit for haven't had time. Have some ssd1306 small and large sizes so can help test.

see3peoh, 2017-06-18:
https://www.bountysource.com/issues/40559111-oled-support-via-i2c

mzahana, 2017-07-10:
Any progress on this?

dagar, 2017-07-10:
It looks fairly straightforward to implement. If someone wants to lend me one and lay out exactly what you want it to do I think we could get basic support together pretty quickly.

see3peoh, 2017-08-23:
I'll send you over one. Messaging you on slack.

dagar, 2017-11-22:
WIP pull request - https://github.com/PX4/Firmware/pull/8336

closed 2018-02-22 by PX4BuildBot

muramura, 2024-01-02:
I would like to implement a driver based on this MIT licensed code.
If I use this code as a reference, can I PR to PX4-Autopilot?

https://github.com/DuyTrandeLion/nrf52-ssd1309