PX4/PX4-Autopilot #6232: Cannot build from source

kanishkanamdeo, 2017-01-02:
I am using Ubuntu 16.04 with Cmake 3.5.2. I am getting the following output when i try to make px4fmu-v2_default

`PX4 CONFIG: /home/kanishka/PX4/src/Firmware/build_px4fmu-v2_default
[  0%] Built target git_gencpp
[  0%] Built target git_genmsg
[  0%] Generating git hash header
[  0%] Built target git_nuttx
[  1%] Built target git_uavcan
[  1%] Built target git_mavlink
[  1%] Built target git_matrix
[  1%] Built target ver_gen
[  1%] Built target git_ecl
[  1%] Built target libuavcan_dsdlc
[  2%] Built target mixer_gen
[  3%] Built target xml_gen
[  4%] Built target nuttx_copy_px4io-v2
[  4%] Built target nuttx_patch_fix_CRTSCTS-define.patch-px4io-v2
[  5%] Built target nuttx_patch_wip_inflight_to_upstream.patch-px4io-v2
[  5%] Built target nuttx_patch_fixStdint.patch-px4io-v2
[  5%] Built target nuttx_copy_px4fmu-v2
[  5%] Built target nuttx_patch_ctype_cctype_.patch-px4io-v2
[  5%] Built target nuttx_patch_nsh_parse.patch-px4io-v2
[  5%] Built target nuttx_patch_math.h.patch-px4io-v2
[  5%] Built target nuttx_patch_export_insitu.patch-px4io-v2
[  6%] Built target nuttx_patch_Fixed-Shadow-wanings.patch-px4io-v2
[  6%] Built target nuttx_patch_silence-jobserver-warnings.patch-px4io-v2
[  6%] Built target nuttx_patch_c++11.patch-px4io-v2
[  6%] Built target nuttx_patch_static_asset-fix.patch-px4io-v2
[  7%] Built target nuttx_patch_serial_dma_hotfix.patch-px4io-v2
[  7%] Built target nuttx_patch_silence-jobserver-warnings.patch-px4fmu-v2
[  7%] Built target nuttx_patch_static_asset-fix.patch-px4fmu-v2
[  7%] Built target nuttx_patch_fixStdint.patch-px4fmu-v2
[  7%] Built target nuttx_patch_serial_dma_hotfix.patch-px4fmu-v2
[  8%] Built target nuttx_patch_export_insitu.patch-px4fmu-v2
[  8%] Built target nuttx_patch_c++11.patch-px4fmu-v2
[  8%] Built target nuttx_patch_Fixed-Shadow-wanings.patch-px4fmu-v2
[  9%] Built target nuttx_patch_fix_CRTSCTS-define.patch-px4fmu-v2
[  9%] Built target nuttx_patch_math.h.patch-px4fmu-v2
[  9%] Built target nuttx_patch_nsh_parse.patch-px4fmu-v2
[  9%] Built target nuttx_patch_wip_inflight_to_upstream.patch-px4fmu-v2
[  9%] Built target nuttx_patch_px4io-v2
[  9%] Built target nuttx_patch_ctype_cctype_.patch-px4fmu-v2
[  9%] Built target nuttx_export_px4io-v2
[  9%] Built target nuttx_patch_px4fmu-v2
[  9%] Built target io_prebuild_targets
[  9%] Built target nuttx_export_px4fmu-v2
[  9%] Built target prebuild_targets
[ 27%] Built target msg_gen
[ 27%] Built target platforms__common
[ 27%] Building CXX object src/platforms/nuttx/CMakeFiles/platforms__nuttx.dir/px4_nuttx_impl.cpp.obj
[ 29%] Built target drivers__device
[ 30%] Built target drivers__stm32
[ 36%] Built target px4io-v2
[ 36%] Built target drivers__stm32__tone_alarm
[ 37%] Built target drivers__stm32__adc
[ 37%] Built target drivers__led
[ 37%] Built target drivers__px4fmu
[ 38%] Built target drivers__rgbled
[ 39%] Built target drivers__px4io
[ 41%] Built target drivers__boards__px4fmu-v2
Scanning dependencies of target drivers__lsm303d
[ 41%] Built target drivers__mpu6000
Scanning dependencies of target drivers__l3gd20
In file included from /usr/arm-none-eabi/include/c++/6.2.1/string:52:0,
                 from /usr/arm-none-eabi/include/c++/6.2.1/stdexcept:39,
                 from /usr/arm-none-eabi/include/c++/6.2.1/array:39,
                 from /usr/arm-none-eabi/include/c++/6.2.1/tuple:39,
                 from /usr/arm-none-eabi/include/c++/6.2.1/functional:55,
                 from /home/kanishka/PX4/src/Firmware/src/include/../platforms/px4_subscriber.h:42,
                 from /home/kanishka/PX4/src/Firmware/src/include/../platforms/px4_nodehandle.h:42,
                 from /home/kanishka/PX4/src/Firmware/src/include/px4.h:46,
                 from /home/kanishka/PX4/src/Firmware/src/platforms/nuttx/px4_nuttx_impl.cpp:40:
/usr/arm-none-eabi/include/c++/6.2.1/bits/basic_string.h: In function 'float std::__cxx11::stof(const string&, std::size_t*)':
/usr/arm-none-eabi/include/c++/6.2.1/bits/basic_string.h:5439:31: error: 'strtof' is not a member of 'std'
   { return __gnu_cxx::__stoa(&std::strtof, "stof", __str.c_str(), __idx); }
                               ^~~
compilation terminated due to -Wfatal-errors.
[ 43%] Built target drivers__mpu9250
Scanning dependencies of target drivers__hmc5883
src/platforms/nuttx/CMakeFiles/platforms__nuttx.dir/build.make:62: recipe for target 'src/platforms/nuttx/CMakeFiles/platforms__nuttx.dir/px4_nuttx_impl.cpp.obj' failed
make[3]: *** [src/platforms/nuttx/CMakeFiles/platforms__nuttx.dir/px4_nuttx_impl.cpp.obj] Error 1
[ 43%] Building CXX object src/drivers/lsm303d/CMakeFiles/drivers__lsm303d.dir/lsm303d.cpp.obj
CMakeFiles/Makefile2:6249: recipe for target 'src/platforms/nuttx/CMakeFiles/platforms__nuttx.dir/all' failed
make[2]: *** [src/platforms/nuttx/CMakeFiles/platforms__nuttx.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 43%] Building CXX object src/drivers/l3gd20/CMakeFiles/drivers__l3gd20.dir/l3gd20.cpp.obj
[ 44%] Building CXX object src/drivers/hmc5883/CMakeFiles/drivers__hmc5883.dir/hmc5883_spi.cpp.obj
[ 44%] Building CXX object src/drivers/hmc5883/CMakeFiles/drivers__hmc5883.dir/hmc5883_i2c.cpp.obj
[ 44%] Building CXX object src/drivers/hmc5883/CMakeFiles/drivers__hmc5883.dir/hmc5883.cpp.obj
[ 44%] Linking CXX static library libdrivers__l3gd20.a
[ 44%] Built target drivers__l3gd20
[ 44%] Linking CXX static library libdrivers__lsm303d.a
[ 44%] Built target drivers__lsm303d
[ 44%] Linking CXX static library libdrivers__hmc5883.a
[ 44%] Built target drivers__hmc5883
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2
`

mhkabir, 2017-01-02:
What is your ARM-GCC version?

kanishkanamdeo, 2017-01-02:
@mhkabir  gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609

mhkabir, 2017-01-02:
Is that from `arm-none-eabi-gcc --version` ?

julianoes, 2017-01-02:
@kanishkanamdeo looks more like 6.2.1:
```
arm-none-eabi/include/c++/6.2.1
```

Ubuntu recently updated to 6.2.1, please install 5.4 as explained in http://dev.px4.io/starting-installing-linux-boutique.html#toolchain-installation

SDroneS, 2017-01-03:
I have the same problem. I've executed all the commands from [http://dev.px4.io/starting-installing-linux-boutique.html#toolchain-installation](url) to install 5.4 but it didn't helped. `arm-none-eabi-gcc --version` says that it's still 6.2.1

> arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

LorenzMeier, 2017-01-03:
Fixed on master.

SDroneS, 2017-01-03:
It solved previous problem but right now another occurs.

> Traceback (most recent call last):
  File "/home/sds/Repos/Firmware/cmake/nuttx/bin_to_obj.py", line 56, in <module>
    locals())
  File "/home/sds/Repos/Firmware/cmake/nuttx/bin_to_obj.py", line 47, in run_cmd
    raise RuntimeError(stderr)
RuntimeError: /usr/bin/arm-none-eabi-ld: romfs.bin: not in ELF format

> src/firmware/nuttx/CMakeFiles/romfs.dir/build.make:193: recipe for target 'src/firmware/nuttx/romfs.o' failed
make[3]: *** [src/firmware/nuttx/romfs.o] Error 1
CMakeFiles/Makefile2:6544: recipe for target 'src/firmware/nuttx/CMakeFiles/romfs.dir/all' failed
make[2]: *** [src/firmware/nuttx/CMakeFiles/romfs.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

Before the `arm-none-eabi-gcc` update I was able to build the code smoothly.

julianoes, 2017-01-03:
@SDroneS you need to make sure your `$PATH` is correct and the installed `arm-none-eabi-*` is actually chosen. Also `which arm-none-eabi-gcc` will tell you which one it uses.

SDroneS, 2017-01-03:
I have figured that out, but no matter what `arm-none-eabi-gcc` version I use (tried 4.9, 5.4 and current 6.2.1), errors from above exist. It's the same issue as [#6172](https://github.com/PX4/Firmware/issues/6172) despite it's supposed to be fixed.

LorenzMeier, 2017-01-03:
Make sure to run `make clean` and if that doesn't help `make distclean`. 4.9 and 5.4 definitely work, and 6.x got fixed so "the issue" is definitely fixed. You have a local issue with your setup, which is also apparent from having a different error message.

kd0aij, 2017-01-04:
@LorenzMeier I just reproduced the "not in ELF format" error under Ubuntu 16.04 with gcc6.
@SDroneS Did you install 32 bit support libraries?
Try this:
sudo apt-get install libc6:i386 libgcc1:i386 libstdc++5:i386 libstdc++6:i386
(note that gcc-4.6-base:i386 is no longer available)
and then try gcc5.4: worked for me.

dagar, 2017-01-04:
Might be a bad idea, but we could maintain scripts in the Tools folder to install all dependencies for each supported platform. Ubuntu LTS (14.04 and 16.04), OSX, maybe Fedora if there's interest. This could be the one true version that we use in the dev guide, docker image creation, and refer people to. The reality is it will constantly change, but this would concentrate the effort.

SDroneS, 2017-01-04:
Finally works! 
@kd0aij You were right, some of 32 bit support libraries were missing. I didn't have to install them in the past (I mean before the update to 6.2.1). Right now I'm able to build with 5.4 but when it comes to 6.2.1 I still get what's posted above. Maybe I do something wrong, doesn't matter.
@dagar Great idea. There's a lot of confusion right now.

julianoes, 2017-01-04:
I get the same for 6.2.1:
```
Traceback (most recent call last):
  File "/home/julianoes/src/Firmware/cmake/nuttx/bin_to_obj.py", line 56, in <module>
    locals())
  File "/home/julianoes/src/Firmware/cmake/nuttx/bin_to_obj.py", line 47, in run_cmd
    raise RuntimeError(stderr)
RuntimeError: /usr/bin/arm-none-eabi-ld: romfs.bin: not in ELF format

src/firmware/nuttx/CMakeFiles/romfs.dir/build.make:76: recipe for target 'src/firmware/nuttx/romfs.o' failed
```

kd0aij, 2017-01-04:
@dagar I edited the advanced linux toolchain installation guide to make the 32 bit lib installation more prominent, and separated the problem package from the ones which installed for me on Ubuntu 16.04

ivan95603, 2017-01-05:
Thanks @kd0aij,
Done:

pushd .
cd ~
wget https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
tar -jxf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
exportline="export PATH=$HOME/gcc-arm-none-eabi-5_4-2016q2/bin:$PATH"
if grep -Fxq "$exportline" ~/.profile; then echo nothing to do ; else echo $exportline >> ~/.profile; fi
. ~/.profile
popd

make clean

make distclean

and it works now with:

arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

kanishkanamdeo, 2017-01-05:
@kd0aij Thank you so much, the builds have resumed, and I got back on the 5.4.1. Yay!

julianoes, 2017-01-05:
@kd0aij I'd say we should change the devguide to propose the manual install, see https://github.com/PX4/Devguide/pull/78.

kanishkanamdeo, 2017-01-05:
@julianoes I second that!

closed 2017-02-16 by dagar