PX4/PX4-Autopilot #6235: Mavlink rate control is still wonky

mhkabir, 2017-01-02:
So I'm trying to validate the 250 Hz data transfer result from https://github.com/PX4/Firmware/pull/6169, but something still isn't right.

I'm using the USB CDC interface on a Pixhawk, starting only a few streams in mavlink custom mode : 

Here's how I'm starting mavlink : 

```
# Mavlink
mavlink start -r 800000 -d /dev/ttyACM0 -m custom -x

# VIO Specific Streams
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU 			-r 200
mavlink stream -d /dev/ttyACM0 -s GPS_RAW_INT 			-r 5
mavlink stream -d /dev/ttyACM0 -s ATTITUDE_QUATERNION 		-r 30
mavlink stream -d /dev/ttyACM0 -s CAMERA_TRIGGER 		-r 1000
mavlink stream -d /dev/ttyACM0 -s TIMESYNC 			-r 30
mavlink stream -d /dev/ttyACM0 -s SYSTEM_TIME 			-r 1

# Debug streams
mavlink stream -d /dev/ttyACM0 -s VISION_POSITION_ESTIMATE 	-r 10
mavlink stream -d /dev/ttyACM0 -s LOCAL_POSITION_NED 		-r 10
mavlink stream -d /dev/ttyACM0 -s GLOBAL_POSITION_INT 		-r 10
mavlink stream -d /dev/ttyACM0 -s SYS_STATUS 			-r 5
mavlink stream -d /dev/ttyACM0 -s EXTENDED_SYS_STATE 		-r 5
mavlink stream -d /dev/ttyACM0 -s HOME_POSITION 		-r 1

mavlink boot_complete
```
Note that setting `-r 800000` for `mavlink start` higher/lower _does NOT_ affect anything, and the issue persists.
Setting the `-r 200` for `mavlink stream` higher (without touching anything else) to e.g 250 _DOES_ increase the rate (to something like 210 Hz) but it never reaches the target rate.

From `mavlink status` :

```
instance #0:
        GCS heartbeat:  1182900777 us ago
        mavlink chan: #0
        type:           USB CDC
        flow control:   OFF
        rates:
        tx: 17.202 kB/s
        txerr: 0.000 kB/s
        rx: 0.996 kB/s
        rate mult: 1.000
        accepting commands: YES
        MAVLink version: 1
        transport protocol: serial (/dev/ttyACM0 @2000000)
``` 

Note that the rate multiplier is 1.0.

And the actual rate I'm getting (for HIGHRES_IMU): 

```
average rate: 186.141
	min: 0.002s max: 0.010s std dev: 0.00172s window: 177
average rate: 184.189
	min: 0.002s max: 0.010s std dev: 0.00197s window: 360
average rate: 184.016
	min: 0.002s max: 0.010s std dev: 0.00203s window: 544
average rate: 184.252
	min: 0.002s max: 0.010s std dev: 0.00199s window: 728
average rate: 184.166
	min: 0.002s max: 0.010s std dev: 0.00197s window: 765
```

@LorenzMeier Your help would be greatly appreciated on this one. Been running into it for ages, and I still don't understand why...  (exactly same as https://github.com/PX4/Firmware/issues/3291).

blandry, 2017-08-09:
@mhkabir do you have any update on this?

mhkabir, 2017-08-09:
This should be fixed on master. There were a bunch of fixes related to scheduling. Let me know if it isn't.

closed 2017-08-09 by mhkabir

blandry, 2017-08-10:
@mhkabir looks like it did yes. Quick question for you, is there anything artificially limiting all the rates to 250Hz? Or were you ever able to get a higher rate for any of the streams Thank you!

mhkabir, 2017-08-10:
@blandry OK the mainloop speed seems to be fine. What message are you trying to send? I'm pretty sure that there is actually no message corresponding to a topic which is updated at more than 250Hz.

blandry, 2017-08-11:
@mhkabir hmm right I think you're correct. Somehow I was under the impression that I could get attitude at 500Hz, but the EKF certainly doesn't run at that rate. Thanks!

LorenzMeier, 2017-08-11:
@blandry It is feasible to run it at that rate - would you like to get some pointers to increase the rate?

blandry, 2017-08-11:
@LorenzMeier actually that would be very helpful yes! I'm worried that messing with the scheduler too much without knowing the limitations of the hardware could lead to unforeseen problems.

gijimbo56, 2018-03-06:
That would be great to learn how to run the EKF at that rate. What are the pointers to do this? Assuming the EKF runs that fast, changing the estimator rates wouldn't necessarily change the update rates to 250Hz or greater as @mhkabir had mentioned, it seems thats regulated by something else in the code? Thanks in advance.