PX4/PX4-Autopilot #6242: LPE: Landing in ALTHOLD mode difficult due to accelerator measurements at touch down

ecmnet, 2017-01-03:
@jgoppert I have a quite soft vibration damping installed to achieve an acceleration noise of about 0.1g - 0.15g in Z-Axis, which seems to work perfectly during flight and gives awesome altitude hold accuracy. But when landing in ALTHOLD, the accelerator measurements are very high (and seem to vibrate) at touch down. As a result the vehicle 'jumps' a bit and would sometimes crash if I do not reduce throttle at once (or use the kill switch).

ACC.Z:

[image]

LPOS.Z and LPOS.VZ:

[image]

I do not want to make the vibration damping harder in my case, but would like to improve the landing experience in ALTHOLD (and POSHOLD). Any idea to make that more stable within LPE? Would it help to introduce a limit on ACC.Z?

whyscience, 2017-01-04:
what's the tool you use to plot ?

ecmnet, 2017-01-04:
Not relevant anymore with increased spring constant in damping

closed 2017-01-04 by ecmnet

mhkabir, 2017-01-04:
@ecmnet, would to mind sharing a picture of your damping solution? :)

ecmnet, 2017-01-04:
@mhkabir Nothing special about it. It's a question of the angles and the foam :-)

[image]

I am still looking for cable wire springs for loads of about 20g.