PX4/PX4-Autopilot #6250: logger support for imu noise characterization

jgoppert, 2017-01-04:
@mhkabir and I are trying to automate noise characterization for the estimators based on log files. The issue is that we currently have to use full rate data to get the correct noise properties which when running for 6 hours for an Allan variance plots results in log files of several GB's even with minimal topic logging. It would be great if the logger had an option for computing the mean of the data over a sampling interval. This would allow us to collect data at 1Hz instead of 250 Hz without degradation in the estimation of the noise properties.

## Full rate sampling

```python
d0 = px4tools.ulog.read_ulog('./logs/19_19_32.ulg', 'sensor_gyro_0')
sensor_gyro = d0.get_topic('sensor_gyro_0')
d0_sensor_gyro.index = pandas.TimedeltaIndex(d0_sensor_gyro.index, unit='s')
px4tools.ulog.plot_allan_variance(d0_sensor_gyro.t_sensor_gyro_0__f_x)
```
[image]

{'sig_bi': 8.0655112895018711e-05,
 'sig_rrw': 7.1345534585751902e-06,
 'sig_rw': 0.00040261397570768773,
 'tau_0': 28.756570159338182,
 'tau_1': 99.821604544821938,
 'tau_2': 295.95871939906078}

## Forward fill based aggregation (what the logger is doing when downsampling)

```python
px4tools.ulog.plot_allan_variance(
    d0_sensor_gyro.resample('1000L').agg('ffill').t_sensor_gyro_0__f_x)
```
[image]

* Note the statistics are different than the full rate sampling
* We could probably correct this if we knew the true logging rate, but it will be difficult for users to know that for every topic.

{'sig_bi': 0.00014868078312748142,
 'sig_rrw': 8.5133948764007763e-06,
 'sig_rw': 0.0013287844678578648,
 'tau_0': 1.7226126313432268,
 'tau_1': 247.95841818123941,
 'tau_2': 655.43931432275701}

## Mean based aggregation (what we want the logger to be able to do)

```python
px4tools.ulog.plot_allan_variance(
    d0_sensor_gyro.resample('1000L').agg('mean').t_sensor_gyro_0__f_x)
```

[image]

{'sig_bi': 8.2815208762766174e-05,
 'sig_rrw': 7.1562621560294453e-06,
 'sig_rw': 0.00040628705706080321,
 'tau_0': 26.383730114503347,
 'tau_1': 101.01425772904228,
 'tau_2': 318.55649477568477}

* Note there is no significant difference in the statistics from the full rate logging.

bkueng, 2017-01-04:
I suggest to do this in another module which publishes a separate topic at 1Hz. Rather than adding it directly to the logger which has been designed to be ignorant of topic content.

jgoppert, 2017-01-07:
I agree. It makes sense to have a secondary sensor combined topic that holds the means of the raw sensors over the last second, controlled by a separate px4 module that subscribes to the raw sensor data.

closed 2018-02-21 by PX4BuildBot