PX4/PX4-Autopilot #6258: Can't build master branch Ubuntu 16.04 LTS says romfs.bin: not in ELF format

ivan95603, 2017-01-04:
make px4fmu-v2_default
-- nuttx-px4fmu-v2-default
-- CMAKE_INSTALL_PREFIX: 
-- cmake module path: /home/ivan/src/Firmware/cmake
-- Nuttx build for px4fmu-v2 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- NuttX patch: nuttx-patches/Fixed-Shadow-wanings.patch
-- NuttX patch: nuttx-patches/c++11.patch
-- NuttX patch: nuttx-patches/ctype_cctype_.patch
-- NuttX patch: nuttx-patches/export_insitu.patch
-- NuttX patch: nuttx-patches/fixStdint.patch
-- NuttX patch: nuttx-patches/fix_CRTSCTS-define.patch
-- NuttX patch: nuttx-patches/math.h.patch
-- NuttX patch: nuttx-patches/nsh_parse.patch
-- NuttX patch: nuttx-patches/serial_dma_hotfix.patch
-- NuttX patch: nuttx-patches/silence-jobserver-warnings.patch
-- NuttX patch: nuttx-patches/static_asset-fix.patch
-- NuttX patch: nuttx-patches/wip_inflight_to_upstream.patch
-- Using C++03
-- Release build type: RelWithDebInfo
-- Adding UAVCAN STM32 platform driver
-- Adding ROMFS on px4fmu-v2
-- Nuttx build for px4fmu-v2 on m3 hardware, using nsh 
-- NuttX patch: nuttx-patches/Fixed-Shadow-wanings.patch
-- NuttX patch: nuttx-patches/c++11.patch
-- NuttX patch: nuttx-patches/ctype_cctype_.patch
-- NuttX patch: nuttx-patches/export_insitu.patch
-- NuttX patch: nuttx-patches/fixStdint.patch
-- NuttX patch: nuttx-patches/fix_CRTSCTS-define.patch
-- NuttX patch: nuttx-patches/math.h.patch
-- NuttX patch: nuttx-patches/nsh_parse.patch
-- NuttX patch: nuttx-patches/serial_dma_hotfix.patch
-- NuttX patch: nuttx-patches/silence-jobserver-warnings.patch
-- NuttX patch: nuttx-patches/static_asset-fix.patch
-- NuttX patch: nuttx-patches/wip_inflight_to_upstream.patch
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ivan/src/Firmware/build_px4fmu-v2_default
PX4 CONFIG: /home/ivan/src/Firmware/build_px4fmu-v2_default
Scanning dependencies of target ver_gen
Scanning dependencies of target git_genmsg
Scanning dependencies of target git_gencpp
Scanning dependencies of target git_nuttx
[  0%] Generating git_init_Tools_gencpp.stamp
[  0%] Generating git_init_Tools_genmsg.stamp
[  0%] Generating git_init_NuttX.stamp
[  0%] Built target git_gencpp
[  0%] Built target git_genmsg
[  0%] Built target git_nuttx
[  0%] Generating git hash header
Scanning dependencies of target git_uavcan
Scanning dependencies of target git_matrix
Scanning dependencies of target git_mavlink
[  0%] Generating git_init_src_lib_matrix.stamp
[  1%] Generating git_init_src_modules_uavcan_libuavcan.stamp
[  1%] Generating git_init_mavlink_include_mavlink_v1.0.stamp
[  1%] Built target git_matrix
[  1%] Built target git_uavcan
[  1%] Built target git_mavlink
Scanning dependencies of target mixer_gen
[  1%] Built target ver_gen
[  2%] Generating mixer_multirotor.generated.h
Scanning dependencies of target git_ecl
Scanning dependencies of target libuavcan_dsdlc
[  2%] Generating git_init_src_lib_ecl.stamp
[  2%] Running dsdl compiler
[  2%] Built target mixer_gen
[  2%] Built target git_ecl
[... 751 lines cut ...]
[ 98%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/sensors/mag.cpp.obj
[ 98%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/sensors/baro.cpp.obj
[ 98%] Linking CXX static library libmodules__uavcan.a
[ 98%] Built target modules__uavcan
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cmake --version
cmake version 3.5.2

gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609

-------------------------------------------------------------------------------------------------------------------
Tried as LorenzMeier said make clean and make distclean and also he said that the gcc 6.2.1 was fixed on master but i can't build. 
Thanks
-------------------------------------------------------------------------------------------------------------------

dagar, 2017-01-04:
What about `arm-none-eabi-gcc --version`?

kd0aij, 2017-01-04:
Duplicate issue: see https://github.com/PX4/Firmware/issues/6232#issuecomment-270263524

ivan95603, 2017-01-04:
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]

ivan95603, 2017-01-04:
kd0aij,
It is similar but it hasn't been fixed yet I also tried
sudo apt-get install libc6:i386 libgcc1:i386 libstdc++5:i386 libstdc++6:i386
still nothing

dagar, 2017-01-04:
@ivan95603 as a temporary work around I suggest you switch to arm-none-eabi-gcc 5.4 as described in the dev guide. http://dev.px4.io/starting-installing-linux-boutique.html

Follow https://github.com/PX4/Firmware/issues/6232 to see when 6.2 is working.

closed 2017-01-04 by dagar

kd0aij, 2017-01-04:
use the 5.4 toolchain, not 6.2, as it says in the comment

ivan95603, 2017-01-05:
Thanks @kd0aij,
Done:

pushd .
cd ~
wget https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
tar -jxf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
exportline="export PATH=$HOME/gcc-arm-none-eabi-5_4-2016q2/bin:\$PATH"
if grep -Fxq "$exportline" ~/.profile; then echo nothing to do ; else echo $exportline >> ~/.profile; fi
. ~/.profile
popd

**make clean**

**make distclean**

and it works now with:

**arm-none-eabi-gcc --version**
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]