PX4/PX4-Autopilot #6259: TeraRangerOne fails to start

mzahana, 2017-01-05:
Hi,

I am not able to run TeraRAngerOne with later PX4 firmware on Pixhawk 2.1.
1- I connected TeraRangerOne to Pixhawk2.1 I2C
2- I change the 'trone' parameter in QGC to 'Enable'
3- I restart the Pixhawk, but the I cannot see the sensor data in QGC
4- I use ```nsh> trone start```
5- it fails with the following error: Trone driver start failed'

What is missing here?

Do I have to solder the JP1 on the I2C adapter?
the TeraRanger sensor version is 1601.

Thanks

ndepal, 2017-01-05:
Does the TROne work on other devices (e.g. via USB on your PC)?
When powering the Pixhawk, do you see the yellow LED on the TROne board flash once?
Is it possible that the I2C address has been changed on this device?
If you're familiar with building the firmware youself, I'd suggest adding some print statements (`warnx`) in the start routine to narrow down the source of the problem: https://github.com/PX4/Firmware/blob/b6b889260689cb340d6815fc68aea831751f48bc/src/drivers/trone/trone.cpp#L732

mzahana, 2017-01-05:
> Does the TROne work on other devices (e.g. via USB on your PC)?
Yes

> When powering the Pixhawk, do you see the yellow LED on the TROne board flash once?
Yes

> Is it possible that the I2C address has been changed on this device?
How/where can I check that? datasheet? I wil double check on that.

I am familiar with building the firmware. I will check on that.  what are the places where [warnx] tags would be useful?

Thanks

ndepal, 2017-01-05:
A different I2C address is unlikely. Older versions of the TROne did not ship with I2C capabilities by default, so you had to request it from the manufacturer, and you could request a specific I2C address. Unless you asked for an alternate firmware with a changed I2C address, this is probably not the cause.

I would just place them in the various failure checks to see which part of the initialization process reports an error.

mzahana, 2017-01-05:
Thanks. We did not ask for special mods on the sensor. The sensor version is 1601.

I placed several checks in the code, and here where it gets stuck:
https://github.com/PX4/Firmware/blob/b6b889260689cb340d6815fc68aea831751f48bc/src/drivers/trone/trone.cpp#L280

```cpp
	if (I2C::init() != OK) {
		goto out;
}
``

mzahana, 2017-01-05:
So, I did more debugging, and it fails here in the ```i2c_nuttx.cpp```:
https://github.com/PX4/Firmware/blob/b6b889260689cb340d6815fc68aea831751f48bc/src/drivers/device/i2c_nuttx.cpp#L151

What could be the issue?

msadowski, 2017-01-12:
@mzahana: Are you sure that your i2c wiring is correct? Check the attached pictures for the wiring (note that the wires on extreme positions are swapped). Otherwise can you check the continuity in all the i2c wires?
[image]

mzahana, 2017-01-12:
Problem is solved. It was a combination of corrupted i2c cable and the i2c
adapter!

closed 2017-01-25 by mzahana