PX4/PX4-Autopilot #6260: MavLink and Logs missing Data

tops4u, 2017-01-05:
Since 1.5.3 I have observed some hickups in the Logs (check: http://discuss.px4.io/t/various-issues-with-1-5-2-stable/2039/7) as well as a meesage from the OSD indicating that the Mavlink Stream has not send data for some time (Message : "Stream Frozen") for OSD with Firmware : https://github.com/night-ghost/minimosd-extra running on 57'600 Baud default Settings for FMUv2 on same Serial as Telemetry. This has been working prior to 1.5.2. 

Stream Frozen would occure in the actual OSD FW when there is no GPS Data in the MavLink Stream for more than 10 Cycles. In earlier Versions the Threshold was even as low as 5 Cycles and I have never seen this meesage. 

I suspect a common Problem for the low Log Rate and the Frozen MAVLINK Stream.

tops4u, 2017-01-14:
@LorenzMeier any Ideas on this one?

tops4u, 2017-02-04:
@LorenzMeier this is still happening. Even a bit more excessive than before. Today I had a few seconds a frozen MavLink Stream on the MinimOSD. Any Hint about the cause? I did not see drops in the FC logs...

LorenzMeier, 2017-03-13:
We don't see that as a common issue so far. Can you try to

  * Use a high-quality SD card
  * Format it if not already done

If you can reproduce this on the bench, run `top` with the serial console connected.

tops4u, 2017-03-13:
@LorenzMeier I use a geniune SanDisk Ultra 16GB Card... I do not think that this is a SD Card Speed related issue since also the MavLink Stream seems to be affected when the OSD does not get MavLink Messages for some time. This seems to have worsened. As written above I have never seen gaps in the Logs nor the MAVLink Stream Frozen Message on the OSD prior to 1.5.2.

I could try to refromat the Card. I have switched to the new uLog Format, but did not recheck if the Problem disappeard since. What is the estimated needed Speed (Latency could be a Problem as well). I think I got about 45MB of Log for 20 min Flight this sums up to no more than 50kB/s that should be no problem even for a cheap Card...

tops4u, 2017-04-02:
@LorenzMeier  / @bkueng  / @davids5 

This got a lot worse now... 

Using current Master I had OSD Stream Dropouts of several Seconds. This is anoying to fly to a certain altitude or distance. Today I could fly 100m (@PosCtl with 20kmh) without any Updates of OSD Data (this is about 18 Secs of Frozen OSD Data Stream). 

This can also be seen as a huge Logging Dropout of the same amount. To be noted there is a also a 433MHz Telemetry Link on the same Serial Port which most probably lost Heartbeat Connection due to the distance. Maybe this has some influence. 

See attached Log which reflects the dropout.

http://logs.px4.io/plot_app?log=198306da-b5d3-4da9-b733-5124a8f4b7fe

LorenzMeier, 2017-04-02:
I think this is related to the trigger setup. We don't see this issue in general.

tops4u, 2017-04-02:
@LorenzMeier the Camera Trigger? This should be inactive, or am I wrong? 

I would like to be able to fire the Camera Trigger (GPIO) via RC or QGC. 

What can I do to check?

tops4u, 2017-04-03:
Just one more note: I did not trigger the camera, since I was using Stable QGC until yesterday which did not offer to trigger the Camera.

dagar, 2017-04-11:
So this is actual telemetry dropping during this period?

tops4u, 2017-04-12:
@dagar : I did not check the Data on QGC during this time but I would assume so as the OSD and the Telemetry are sharing the same Serial Port.

tops4u, 2017-05-11:
This Problem is still persisting even after reverting Camera Trigger Setup to default. 
IMHO it might be less often now but still happening.

dagar, 2017-05-16:
Have you seen it on multiple autopilots? Please keep positing logs when it happens if possible.

tops4u, 2017-05-16:
@dagar Thanks for taking care of my Problem. I just updated yesterday to the actual Master and I still have the Problem of the Frozen Stream Indication on the OSD. I see various possible causes for this:
- Serial Port congestion due to 3DR Radio and OSD sharing the same Port
- Sporadic electrical Interference on the Serial Line
- OSD Logic BUG
- HW Failure on the Serial Port of my FC

But this might be hard to track down. I can also see sporadic delays in Log writing lasting from a few uS to about 15-20 secs. Latest Log from today was 6sec (but just before Taking off)
http://logs.px4.io/plot_app?log=1b11172e-9917-4f73-90e8-465048e9ffc9

Anything I could perform on the shell when this happens?

tops4u, 2017-05-30:
@dagar I upgraded my Pixracer to 1.6.0 RC and now the OSD there shows the same Problem. Did any change to the MAVLINK and/or OSD Handling happen which might confuse the OSD Logic? 
I have noted that even when there is no Stream Frozen Message the Altitude is updated only sporadically - which is very anoying when I try to reach a certain Altitude I always under- or overshoot. 

I use this Firmware on the MinimOSD: https://github.com/night-ghost/minimosd-extra 

Do you have any prefered OSD Operating Firmware that works well with PX4?

dagar, 2017-05-30:
No, but I personally don't use OSDs anymore.

1. Have you reproduced this on more than one autopilot?

2. Can you reproduce it if you only connect an OSD and nothing else?

tops4u, 2017-05-30:
@dagar

1. Yes on a Pixracer and on AUAV-X2

2. Yes the Pixracer has no Telemetry only the OSD.

tops4u, 2017-06-15:
I'll be checking this soon with a UAVPlay OSD to see if this also behaves like this, or if this is a sole Problem of the MinimOSD or its Firmware. Don't put any efforts on this for the time being.

tops4u, 2017-11-07:
I suspect a Problem with the OSD or OSD Firmware. I'll close this.

closed 2017-11-07 by tops4u