PX4/PX4-Autopilot #6261: Snapdragon muorb parameter update

Stifael, 2017-01-05:
When doing a parameter update from QGC, the parameter is correctly updated on the krait side, but not on the dsp side. In fact, after a parameter update the value shown on the dsp is equal to the previous value from krait.
In order to reproduce this error: 
1.) choose a parameter such as MC_PITCHRATE_P
2.) open mc_att_control.cpp and do 	PX4_INFO("pitch rate: %.5f",(double)_params.rate_p(1)) somewhere in the code
3.) start px4 and open mini-dm
4.) open QGC
5.) change the parameter MC_PITCHRATE_P to a new value
-> observe the value that is printed in the mini-dm: it will show the previous value
6.) on krait side type: param show MC_PITCHRATE_P
-> it will show correctly the new updated value

mcharleb, 2017-01-09:
@Stifael https://github.com/PX4/Firmware/pull/5327 may fix this issue

Stifael, 2017-01-10:
@mcharleb:
unfortunately #5327 did not fix it, but now I cant even update anything from QGC due to shmem error:
[image].
I also checked with @ChristophTobler and he has the same error.

mcharleb, 2017-01-10:
@Stifael I'll debug it now

mcharleb, 2017-01-11:
@Sitfael I fixed the param output for qurt builds in https://github.com/PX4/Firmware/pull/6297

If you start PX4 and show the parameters:

pxh> param show SDLOG_EXT
Symbols: x = used, + = saved, * = unsaved
x + SDLOG_EXT [819,819] : 2

 928 parameters total, 928 used.

pxh> qshell param show SDLOG_EXT

mini-dm shows:
[08500/00]  41:39.241  HAP:55:Symbols: x = used, + = saved, * = unsaved   0344  param.c
[08500/00]  41:39.241  HAP:55:x + SDLOG_EXT [819,819] :   0457  param.c
[08500/00]  41:39.241  HAP:55:2   0466  param.c
[08500/00]  41:39.241  HAP:55:  928 parameters total, 928 used.   0346  param.c

There are extra line breaks in the mini-dm output but at least you can see the parameter updates.

If I set a parameter on ARM side it is updated on the DSP and vise versa .

mcharleb, 2017-01-11:
@Stifael I was able to build PX4, and then run it on the SD Flight board and set SDLOG_RATE to 10 via QGC and saw that the updated setting was present:

[08500/00]  30:22.288  HAP:54:Symbols: x = used, + = saved, * = unsaved   0344  param.c
[08500/00]  30:22.290  HAP:54:x + SDLOG_RATE [823,823] :   0457  param.c
[08500/00]  30:22.290  HAP:54:10   0466  param.c <<<-- Set to 10
[08500/00]  30:22.290  HAP:54:  928 parameters total, 928 used.   0346  param.c

LorenzMeier, 2017-01-12:
@Stifael still present?

ChristophTobler, 2017-01-12:
I tested it with the branch ```param_qurt_printf_fix``` and in the mini-dm output it still shows:
```sh
[08500/00]  05:13.756  HAP:28712:Could not get lock, file name: /shmem_qurt.c, line number: 217.   0113  shmem_qurt.c
[08500/03]  05:13.756  HAP:28712:Could not get shmem lock   0218  shmem_qurt.c
[08500/00]  05:13.860  HAP:28712:Could not get lock, file name: /shmem_qurt.c, line number: 217.   0113  shmem_qurt.c
[08500/03]  05:13.860  HAP:28712:Could not get shmem lock   0218  shmem_qurt.c
```

And when I change a parameter in QGC I still get
```sh
______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [sdlog2] [blackbox] /root/log/2017-01-12
INFO  [sdlog2] [blackbox] recording: 06_53_26.px4log
INFO  [dataman] Unkown restart, data manager file '/home/linaro/dataman' size is 47640 bytes
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
Sleeping for 1 s; (1000000 us).
pxh> INFO  [mavlink] using network interface wlan0, IP: 10.20.9.140
INFO  [mavlink] with broadcast IP: 10.20.15.255
INFO  [mavlink] partner IP: 10.20.9.8
ERROR [platforms__posix__px4_layer] update_value_from_shmem get param failed
ERROR [platforms__posix__px4_layer] update_value_from_shmem get param failed
ERROR [platforms__posix__px4_layer] update_value_from_shmem get param failed
ERROR [platforms__posix__px4_layer] krait update param 421 failed
```
But I'll wait for @Stifael to confirm

Stifael, 2017-01-12:
Hm, I still have the same issue. In fact, even if I dont update the param, after some time I will get the error "Could not get lock, ...". Also, when I do ctrl c to shut down px4, I receive the error: [muorb] ERROR: Getting Bulk data from fastRPC link.

Maybe we are not using the newest adsp firmware. I will share the files that we used for adsp, linux image and hexagon sdk, so maybe we need to have newer versions.

LorenzMeier, 2017-01-12:
It's worth doing an update to latest anyway. This looks like the system is not configured properly on a very fundamental level.

Stifael, 2017-01-12:
@mcharleb: can you share with me and @ChristophTobler  what BSP image you used? the one I use (Flight_3.1.1_BSP_apq8074-00003.zip) is not on Intrinsyc anymore. For the adsp firmware I used the qcom_flight_controller_hexagon_sdk_add_on.zip.

LorenzMeier, 2017-01-12:
If its not online any more its not even worth investigating further. Please update the board and re-test. This field is moving too quickly to spend time on stale software.

Stifael, 2017-01-12:
Current status: 
I updated two snapdragons with newest linux image and ADSP firmware. Snapdragon_A is the one I used to work with. Snapdragon_B is one that I took from an old vehicle and is older. Snapdragon_C is the snapdragon from @julianoes  with a older linux image but same ADSP firmware. Snapdragon_D is another, newer snapdragon with same state as Snapdragon_C.

Snapdragon_A: 
I still have the same error. 
Error1: It continuously prints error messages that mag sensor initialization failed. It is ok to receive this message one, but I dont know why it sends it continuously with a high frequency. 
[image]
Error2: After some time or if I update a parameter (it occurs very random), I receive the shmem lock error.

Snapdragon_B:
Error1: 
Still remains, but with a much lower frequency.
[image]

Error2:
I did not get it, which is good.

Snapdragon_C:
Error1:
Did not get it. It only prints the error message once.
Error2.
Did also not get it.

Snapdragon_D:
Error1 and Error2 present with high frequency error message update.

IMHO, I think the two errors might be related because the higher the frequency of the error message, the more likely it is to get the shmem lock error ("Could not get lock"). It is also possible that Error1 just uncovers Error2.
I dont know why Error1 occurs with different frequency, and that is maybe something I can look into. Any suggestions?

mcharleb, 2017-01-12:
@LorenzMeier I have reproduced the issue when I use the 3.1.2 release from Intrinsyc. When I use an internal release, I am not seeing the shmem issue. I am investigating.

Stifael, 2017-01-16:
#6312 fixes the shemlock error. Also can confirm that the parameter updates are working now as well.
The is one more issue still: the continuously message sending, but thats a different issue and therefore this issue can be closed.

closed 2017-01-16 by Stifael