PX4/PX4-Autopilot #6263: Parameter ... not found on 272f1dd with Snapdragon

genemerewether, 2017-01-05:
- All calls to `param_get()` fail
- `param show *` displays no parameters
- Tried deleting params file with no change

Git bisect reveals [6fe9b8](https://github.com/PX4/Firmware/commit/6fe9b8e54351c8fe620f959b62ce1902b514f3ab) as the first bad commit

Tested with `make distclean` before building each commit

bkueng, 2017-01-06:
Same problem for `make posix_sitl_ekf2`. The problem is that https://github.com/PX4/Firmware/blob/master/cmake/configs/posix_sitl_ekf2.cmake uses an include for the module list which is not properly handled by the CMakeParser.

@mazahner can you look into that?

bkueng, 2017-01-06:
It's fixed now with https://github.com/PX4/Firmware/pull/6267. Leaving this open as it will need a proper fix.

mazahner, 2017-01-06:
I have not modified the CMakeParser which has been used to generate the px4_parameters.h/.cpp i just reused it during the generation of the parameters.xml (and therefore put it into it's own file). 
So if this has been working before, i wonder how the px4_paramters.h/.cpp have been generated before. Or does the sitl environment only rely on the .xml file?

I'll see what i can do...

bkueng, 2017-01-06:
I know. The problem was already there before, but it did not matter because the scope was not used. So you don't have to, but would be cool if you looked into it anyway.

mhkabir, 2017-04-03:
Is this still an issue?

bkueng, 2017-04-03:
No, we now use cmake directly to get the list of modules instead of parsing it with python.

closed 2017-04-03 by bkueng