PX4/PX4-Autopilot #6266: Please change macro definition to fix native build on raspberry pi

crossa, 2017-01-06:
px4/Firmware/src/lib/DriverFramework/framework/include/

on raspberry pi b+ ,mpu9250 spi device 0  is  /dev/spidev0.0
and mag should be /dev/spidev0.0

`#if defined(__IMU_USE_I2C)
#include "I2CDevObj.hpp"
#else
#include "SPIDevObj.hpp"
#endif

#if defined(__DF_QURT)
#include "dev_fs_lib_spi.h"
#define IMU_DEVICE_PATH "/dev/spi-1"
#elif defined(__DF_BEBOP)
#define IMU_DEVICE_PATH "/dev/i2c-mpu6050"
#elif (defined(__DF_RPI) && !defined(__IMU_USE_I2C))
#define IMU_DEVICE_PATH "/dev/spidev0.0"
#elif (defined(__DF_RPI) && defined(__IMU_USE_I2C))
#define IMU_DEVICE_PATH "/dev/i2c-1"
#elif defined(__DF_EDISON)
#define IMU_DEVICE_PATH "/dev/spidev5.1"
#else
#define IMU_DEVICE_PATH "/dev/spidev0.0"
#endif

#if (defined(__DF_RPI) && !defined(__IMU_USE_I2C))
#include <linux/spi/spidev.h>
#define IMU_DEVICE_ACC_GYRO "/dev/spidev0.0"
#define IMU_DEVICE_MAG "/dev/spidev0.0"
#elif  (defined(__DF_RPI) && defined(__IMU_USE_I2C))
#include <linux/i2c-dev.h>
#define IMU_DEVICE_ACC_GYRO "/dev/i2c-1"
#define IMU_DEVICE_MAG "/dev/i2c-1"
#else
#define IMU_DEVICE_ACC_GYRO ""
#define IMU_DEVICE_MAG ""
#endif

#define IMU_CLASS_PATH  "/dev/imu"`

dagar, 2017-01-06:
@crossa could you submit a pull request?

PX4 DriverFramework is here - https://github.com/PX4/DriverFramework

bkueng, 2017-01-12:
@crossa according to upstream Navio2 [0], the devices are correct. Are you using an older Navio board?

Also, Navio2 only supports RPi 2+3 [1].

[0] https://github.com/emlid/Navio2/blob/navio2/C++/Navio/LSM9DS1.cpp#L7
[1] https://docs.emlid.com/navio2/Navio-APM/hardware-setup/

crossa, 2017-02-14:
OK，I had fix it without any change。I try to build a coptor with pi zero without navio or pxfmini

I’ll  submit a pull request when I finish it

bkueng, 2017-02-14:
Ok, good, closing this then.

closed 2017-02-14 by bkueng