PX4/PX4-Autopilot #6281: mixer test intermittent failure

dagar, 2017-01-09:
@LorenzMeier @AndreasAntener @bkueng 

I thought the mixer test failing was a result of the code coverage build (-O0), but it seems decreasing the optimization level only increases the failure rate. It's still happening under normal settings (make tests).

I can look at it later this week if no one else has time, but I don't know the mixer code (yet).

https://circleci.com/gh/PX4/Firmware/2792?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

	INFO  [tests] RUNNING TEST: loadIOPass
	INFO  [tests] TEST PASSED: loadIOPass
	INFO  [tests] RUNNING TEST: loadQuadTest
	ERROR [modules__unit_test] Compare failed: check number of mixers loaded - (mixer_group.count():4) (expected_count:5) (../src/systemcmds/tests/test_mixer.cpp:353)
	ERROR [tests] Mixer load failed with chunk size 58
	ERROR [tests] TEST FAILED: loadQuadTest
	INFO  [tests] RUNNING TEST: loadVTOL1Test
	ERROR [modules__unit_test] Compare failed: check number of mixers loaded - (mixer_group.count():3) (expected_count:4) (../src/systemcmds/tests/test_mixer.cpp:353)
	ERROR [tests] Mixer load failed with chunk size 61
	ERROR [tests] TEST FAILED: loadVTOL1Test
	INFO  [tests] RUNNING TEST: loadVTOL2Test
	ERROR [modules__unit_test] Compare failed: check number of mixers loaded - (mixer_group.count():5) (expected_count:6) (../src/systemcmds/tests/test_mixer.cpp:353)
	ERROR [tests] Mixer load failed with chunk size 61
	ERROR [tests] TEST FAILED: loadVTOL2Test
	INFO  [tests] RUNNING TEST: loadComplexTest
	INFO  [tests] TEST PASSED: loadComplexTest
	INFO  [tests] RUNNING TEST: loadAllTest
	INFO  [tests] Testing all mixers in ROMFS/px4fmu_common/mixers
	INFO  [tests] TEST PASSED: loadAllTest
	INFO  [tests] RUNNING TEST: mixerTest
	INFO  [tests] TEST PASSED: mixerTest
	ERROR [modules__unit_test] SOME TESTS FAILED
	INFO  [modules__unit_test]   Tests passed :      4
	INFO  [modules__unit_test]   Tests failed :      3
	INFO  [modules__unit_test]   Tested assertions : 4268
	mixer FAILED
	Command 'tests' failed, returned -1

LorenzMeier, 2017-01-18:
I captured now one instance:
```
ERROR [tests] Load of mixer failed, last chunk: M: 1

O: 10000 10000 0 -10000 10000

S: 0 7 10000 10000 0 -10000 10000

, transmitted: 302, text length: 0, resid: 0

ERROR [modules__unit_test] Compare failed: check number of mixers loaded (chunk) - (mixer_group.count():4) (expected_count:5) (../src/systemcmds/tests/test_mixer.cpp:359)

ERROR [tests] Mixer load failed with chunk size 58

ERROR [tests] TEST FAILED: loadQuadTest

INFO  [tests] RUNNING TEST: loadVTOL1Test

ERROR [tests] Load of mixer failed, last chunk: M: 2

O: 10000 10000 0 -10000 10000

S: 1 0 10000 10000 0 -10000 10000

S: 1 1 -10000 -10000 0 -10000 10000

, transmitted: 237, text length: 0, resid: 0

ERROR [modules__unit_test] Compare failed: check number of mixers loaded (chunk) - (mixer_group.count():3) (expected_count:4) (../src/systemcmds/tests/test_mixer.cpp:359)

ERROR [tests] Mixer load failed with chunk size 61

ERROR [tests] TEST FAILED: loadVTOL1Test

INFO  [tests] RUNNING TEST: loadVTOL2Test

ERROR [tests] Load of mixer failed, last chunk: M: 2

O: 10000 10000 0 -10000 10000

S: 1 0 -7500 -7500 0 -10000 10000

S: 1 1 -8000 -8000 0 -10000 10000

, transmitted: 436, text length: 0, resid: 0

ERROR [modules__unit_test] Compare failed: check number of mixers loaded (chunk) - (mixer_group.count():5) (expected_count:6) (../src/systemcmds/tests/test_mixer.cpp:359)

ERROR [tests] Mixer load failed with chunk size 61
```

AndreasAntener, 2017-01-18:
Is it possible that https://github.com/PX4/Firmware/pull/6378 helps?

bkueng, 2017-01-18:
It might, but I did not see a clear connection between the 2 (I never reproduced this), so I didn't reference it.

LorenzMeier, 2017-01-20:
Closing this as fixed because symptom and bug match. Will re-open if it appears again.

closed 2017-01-20 by LorenzMeier