PX4/PX4-Autopilot #6286: Add acro mode for fw

CarlOlsson, 2017-01-10:
When setting up a new fw I noticed that there is no acro mode for airplanes according to the [docs](http://dev.px4.io/concept-flight-modes.html).

Is this correct? And in that case how much work would it be to implement?

dagar, 2017-01-12:
Probably not that much work actually, but it's a bit awkward with the way ECL is currently structured. The rate setpoint is converted to bodyrate in control_bodyrate(). https://github.com/PX4/ecl/blob/master/attitude_fw/ecl_roll_controller.cpp#L59

I've been working on a refactor of fw_att_control to use controllib and I was going to throw in acro and rattitude mode. https://github.com/dagar/Firmware/blob/fw_att_control_cleanup/src/modules/fw_att_control/FixedwingAttitudeControl.cpp
Interested in working together to add acro and review/test?

CarlOlsson, 2017-01-12:
Nice! 
I am in the middle of my thesis at the moment so don't have much time left for coding, but I can for sure do my best in reviewing and testing :)

dagar, 2017-01-12:
Do you have any opinions on FW acro? Would you like to see pure rate mode with stick input = bodyrate setpoint for roll, pitch, yaw? We could also do a bit of a hybrid and re-enable attitude control below a certain threshold (rattitude), or hold the current attitude when the sticks are neutral.

CarlOlsson, 2017-01-14:
I would go with pure rate control, I think it is the most fun to fly. At least to start with:)

philippamlong, 2017-01-18:
I am looking for FW acro too. Want to fly FPV and rate mode should give the best experience.

tumbili, 2017-01-23:
@dagar @CarlOlsson I'm also interested in helping drive this. I've spend some time thinking about this and have some comments and questions:

1) We need to handle the turn compensation for acro mode. When you attempt to do a roll the controller will output lots of pitch up which eventually will lead to a yawrate (in earth frame) which then could have a negative (blocking) effect on roll. I've given this a test in JSBSim and had great problems doing a 360 roll when turn compensation was enabled. Maybe we can scale pitch control down when roll angle is close to 90 degrees.

2) We need to think if and how we want to use the rudder during roll maneuvers. Ideally, the rudder could be used to hold pitch during a knife edge maneuver. However, in the simulator I had problems rolling the plane further due to the coupling of the rudder with roll.
I have to say that I didn't use an acro plane in the simulation so this effect might not be as strong as I encountered it.

3) I assume we want to lock the attitude (roll, pitch) when the user is not giving stick inputs?

4) Stick inputs from the user (roll, pitch) should be interpreted as desired body rates, NOT euler angle rates.

@CarlOlsson Carl, does that make sense to you as an experiences acro pilot?

dagar, 2017-01-23:
I was originally thinking of coming back to this a bit later after casually working on the stick_mapper and/or fw_att_control cleanup, but it's easy enough to do now.

I think the main thing is to coordinate and agree on a design here. At a minimum we'll need to refactor ecl to be able to set body rates directly.

1. Should acro even have turn compensation?
2/3. Would that be if sticks are neutral store the current attitude as a setpoint and resume control_attitude? Or just keep the rate 0?
4. yep

If we can work through the details here I'll propose refactoring ecl and the rest of the changes shouldn't be too bad. Do you think we need/want separate rate limits for FW acro?

tumbili, 2017-01-23:
@dagar Comments to
1) Can't really say, I never flew a fixed wing in acro. I could imagine that it could be useful holding the altitude while doing rolls.

3) I would go for keeping the attitude. That may help when flying in turbulence.

dagar, 2017-01-23:
Intuitively I think attitude lock does sound better. It should be easy to experiment with and without as well as per axis. It also easily gives us rattitude from the beginning as well.
I think that should also take care of the knife edge case.

AndreasAntener, 2017-01-23:
There are 2 use cases:
1) fun 3d flying for everybody
2) rate support for planes with properties that make them difficult to fly nice manually (also for good pilots)

For 1) attitude locking is fine, for 2) body rate control is needed. Reading back @CarlOlsson would also go for rate control. I'm also more a fan of 2 because I think if you now how a plane flies fully manual you'd expect a similar behavior in 3D mode and that is not holding attitude.

AndreasAntener, 2017-01-23:
I've quickly looked through a few stabilizers and most of them support both, some of them only with attitude locking. Rate only control is usually advertised as added stability with less adverse impact on agility of the plane.

NX3 Evo: both
FY-40A: attitude
Vector: both
bavarianDemon: both
APM: attitude

dagar, 2017-01-24:
That's for the overview @AndreasAntener 

I think now the most difficult part is deciding how to restructure ECL fw_attitude. The body angular rate transformation is done in control_bodyrate, but depends on the rate setpoint from other controllers from the same iteration.

AndreasAntener, 2017-01-24:
I'll discuss with Roman today but I'm thinking of extracting the PID controls from the control_bodyrate methods and use them directly when in the correct mode. Then split up the block for attitude control. This should work for rate control. For attitude locking I think the ECLs need a mode so you can use them like now but they handle all the axis without the turn compensation. We're probably only going to implement the first part now. Does that make sense to you? Would you want to take over attitude locking?

CarlOlsson, 2017-01-24:
Great to see so much progress here, awesome! I agree with @AndreasAntener, the preferences for acro mode are probably very different from person to person. Some want a mode that does the aerobatic "by itself" and some want a plane that feels more like it is in "assisted manual". However, I think pure rate control is the most natural mode that should be supported. It is also what users outside of the autopilot community would expect when they go with "acro" since it corresponds to normal RC "gyros" e.g. http://www.multiplex-rc.co.uk/multiplex-wingstabi-system-53-c.asp

tumbili, 2017-01-24:
@dagar Here is a video of my first results.
This is controlling body x rate with pitch still controlled as in stabilized. You can see really well how rudder is used to control pitch at large roll angles.

https://www.youtube.com/watch?v=t8Ie6F2PFF4&feature=youtu.be

tumbili, 2017-01-24:
@CarlOlsson I have a question about turning behaviour in acro mode.
When you speak about "full rate control" what do you mean regarding the body_z axis?
When you bank the plane it will try to do a turn which requires non-zero body_z rate. Should there be some kind of turn coordination in acro mode?

CarlOlsson, 2017-01-24:
I would as a start not control rate around body z since it is not really needed to achieve a smoother more wind robust feeling for a plane in acro mode compared to pure manual, it is also not possible to control it for many airframes, e.g. flying wings without rudder.

tumbili, 2017-01-24:
@CarlOlsson Is your plane ready? ;-)
https://www.youtube.com/watch?v=Qz62aW02SPk&feature=youtu.be

CarlOlsson, 2017-01-24:
@tumbili of course :)

AndreasAntener, 2017-02-18:
Implemented in #6567

AndreasAntener, 2017-02-18:
@CarlOlsson you're up ;)

closed 2017-02-18 by AndreasAntener