PX4/PX4-Autopilot #6287: Unable to build source code after fetching PR#4792

aysh-shariff, 2017-01-10:
[image]
[image]

mhkabir, 2017-01-10:
Run `git clean - dfx`

aysh-shariff, 2017-01-10:
Doesn't help. Same errors.

aysh-shariff, 2017-01-10:
[image]
[image]

aysh-shariff, 2017-01-10:
[image]
[image]

kd0aij, 2017-01-10:
what does "git status" show?

aysh-shariff, 2017-01-10:
nothing to commit, working directory clean

kd0aij, 2017-01-10:
that's not the full output, is it?

aysh-shariff, 2017-01-10:
On branch orbit_test
nothing to commit, working directory clean

I fetched PR #4792 and checkedout to orbit_test branch

kd0aij, 2017-01-10:
then try checking the PR branch out directly

aysh-shariff, 2017-01-10:
Tat doesnt fix it.

kd0aij, 2017-01-10:
Strange. That works for me. What might be different about your build environment?

aysh-shariff, 2017-01-10:
I'm not sure what's wrong. I cloned the latest master.... 
Also, I did try manually copying over the changes but tat doesn't work obviously.

kd0aij, 2017-01-10:
what are your OS and toolchain versions?

aysh-shariff, 2017-01-10:
Ubuntu 14.04.1
How do I check the toolchain version ?

kd0aij, 2017-01-10:
arm-none-eabi-gcc --version
The instructions for installation are here: http://dev.px4.io/starting-installing-linux-boutique.html#toolchain-installation

aysh-shariff, 2017-01-10:
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]

kd0aij, 2017-01-10:
that looks OK. Maybe cmake is different? Mine reports:
```
$ cmake --version
cmake version 3.5.1
```

aysh-shariff, 2017-01-10:
cmake version 3.2.2

aysh-shariff, 2017-01-10:
Can tat be a problem ? I am able to build the source code fine except for this PR. Don't want to mess with the build env...

kd0aij, 2017-01-10:
@dagar This is beyond me, Any ideas what's going on here?

dagar, 2017-01-10:
How did you pull #4792? It doesn't merge cleanly with master and needs to be rebased (manually), so I'm guessing you're in a bad state. 
@aysh-shariff could you go through the diff in #4792 and manually reapply it to a new branch of off current master? If you're not comfortable doing that I can later tonight.

aysh-shariff, 2017-01-10:
I guess that's the problem. For manually applying the changes, I did tat for the vehicle_command.msg file but the other file had too many changes. I was short on time so just tried replacing the entire file instead which clearly didn't work.

kd0aij, 2017-01-10:
@aysh-shariff I though all you did was checkout the branch directly from PX4/Firmware

dagar, 2017-01-10:
I'll take a look later. The MC position controller should be functionally the same as it was during that PR, but it was refactored, so it's just a matter of finding where the orbit code now belongs.

@aysh-shariff I'll close this issue now and ping you from #4792 once it's rebased and ready to test.

closed 2017-01-10 by dagar

aysh-shariff, 2017-01-10:
@kd0aij I did tat as well. When tat didn't work I tried manually applying the changes

aysh-shariff, 2017-01-10:
@dagar Tat would be great! 
Thnx!