PX4/PX4-Autopilot #6288: autopilot thinks airplane is upside down

CarlOlsson, 2017-01-10:
When trying to set up a new fixed wing (1.5.3) after calibration the autopilot sometimes thinks it is upside down.
[image]
However when doing all calibrations the images are displayed correctly for accelerometer and compass and the IMU, external compass orientations are all set correctly. This have happened two times and was fixed by resetting all params to defauls and recalibrating everything and rebooting.

kd0aij, 2017-01-10:
Can you reproduce this, then check whether the autopilot orientation parameter is corrupted?

LorenzMeier, 2017-01-11:
Is it possible you got confused when it asked you for the board orientation during the calibration?

CarlOlsson, 2017-01-11:
No I dubble checked the orientations several times and recalibrated everything with the same results. I also checked every time I calibrated mag and accel that the correct images went green in QGC. If It happens again I will force start the logging over NSH.

bartslinger, 2017-01-11:
I vaguely remember I had something similar before, but don't know which version that was. I rebooted the pixhawk and re-did the level-horizon calibration and that fixed it for me.

CarlOlsson, 2017-01-11:
For me it was not possible to do the level horizon calibration since I got a warning message telling me that I had "[excess roll angle](https://github.com/PX4/Firmware/blob/master/src/modules/commander/accelerometer_calibration.cpp#L708)"

dagar, 2017-04-11:
@CarlOlsson did you ever figure out what was going on here? There's not much to go on. I've personally never seen this.

dagar, 2017-04-12:
If you're able to reproduce please open a new issue with logging. Unfortunately there's not much to go on here.

closed 2017-04-12 by dagar

dagar, 2018-06-20:
@tjbordelon can you open a new issue describing your setup and upload a log?