PX4/PX4-Autopilot #6292: How to get lower PWM output?

CarlOlsson, 2017-01-10:
I am probably doing something wrong but here is a [log file](http://logs.uaventure.com/view/R2xAQ5zvnSQBoqDgxf9YAT) (1.5.3) when PWM_MIN, PWM_AUX_MIN and PWM_AUX_DISARMED are all set to 900.
However, as seen in the log and also verified before arming with a PWM reader the output was never below 1000.

Is this intended behavior? And in that case how do you activate the PWM_AUX_DISARMED parameter?

LorenzMeier, 2017-01-11:
If you use a Pixracer there is only the main outputs, so you would need to change those parameters.

CarlOlsson, 2017-01-11:
Ok thanks, but the question remains, how to actually get a lower output than 1000?
I tried again with:
PWM_MIN = 900
PWM_AUX_MIN = 900
PWM_DISARMED = 900
PWM_AUX_DISARMED = 900

When measuring the output on the throttle pin the PWM signal is never lower than 1000. Am I doing something wrong?

LorenzMeier, 2017-01-12:
These values get applied to the outputs selected with the ``set PWM_OUT 1234`` parameter (this is for the caipirinha). So they will only apply to ports 1-4 for this airframe. For other airframes they might apply to less ports. You would need to work with your airframe config and with ```pwm info``` to work out what is not set up correctly.

dagar, 2017-04-11:
I believe this was an airframe issue. Please reopen if I'm wrong.

closed 2017-04-11 by dagar