PX4/PX4-Autopilot #6294: Missing frsky_telemetry in px4fmu-v2-default.px4

samo1234, 2017-01-11:
since version 1.4.4 there is no frsky_telemetry command or binary.
when the option is enabledin SYS_COMPANION system is stuck.

bkueng, 2017-01-11:
This module was disabled [0] due to flash restrictions. You can enable it and disable something else, then compile it yourself. If you have a board that support the full 2MB flash (like pixhawk mini), you can also use the px4fmu-v3_default target.

[0] https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake#L42

closed 2017-02-27 by LorenzMeier

dagar, 2017-02-27:
Hi @samo1234  
frsky_telemetry was temporarily removed from the px4fmu-v2_default build due to flash constraints, but it's back in now.

PR #6687 (2017-02-27): px4fmu-v2 enable ll40ls and frsky_telemetry
 - closes #6686
 - closes #6294

Uses an extra 12KB
commit cff430e1 (2017-02-27): px4fmu-v2 enable ll40ls and frsky_telemetry

 - closes #6686
 - closes #6294