PX4/PX4-Autopilot #6295: clang tidy automatic fixes

dagar, 2017-01-11:
Playing with letting clang-tidy auto fix a few things. Done with `make clang-tidy`.

**readability-braces-around-statements**
https://github.com/PX4/Firmware/compare/master...dagar:readability-braces-around-statements?expand=1

**readability-simplify-boolean-expr**
https://github.com/PX4/Firmware/compare/master...dagar:readability-simplify-boolean-expr?expand=1

**modernize-use-nullptr**
https://github.com/PX4/Firmware/compare/master...dagar:modernize-use-nullptr?expand=1

**modernize-loop-convert**
https://github.com/PX4/Firmware/compare/master...dagar:modernize-loop-convert?expand=1

**modernize-redundant-void-arg**
https://github.com/PX4/Firmware/compare/master...dagar:modernize-redundant-void-arg?expand=1

bkueng, 2017-01-11:
Nice! The 'clang modernize-loop-convert' will need some manual intervention.
Like `for (auto &i : devmap) {` -> `for (const auto &dev : devmap) {`

julianoes, 2017-01-11:
Cool but readability-braces-around-statements should already be covered by astyle.

dagar, 2017-01-11:
@bkueng yes that's better, but it's not like it was const before?

@julianoes I'm not sure why astyle missed these braces. There's also quite a few in the old ekf, but I left it alone.

This was mostly an experiment to see how well these extra clang tools work hands off. There are many checks here that we can gradually enable.
I'll open PRs for the trivial stuff, but what do you guys think about embracing more modern c++ like range based for loops, initialization, less handling of raw pointers, etc?

On that topic cmake currently sets -std=gnu++0x and -std=gnu99 for c. Is that intentional or somewhat arbitrary? Should we set -std=c++11 and -std=c99 or do we actually want gnu extensions?

dagar, 2017-01-14:
Running clang-tidy on travis-ci - #6336
Update .clang-tidy to add more checks.

closed 2017-01-14 by dagar