PX4/PX4-Autopilot #6304: Landing then flip upside down

lovettchris, 2017-01-12:
Weird bug today, trying to land a flamewheel quadrocopter using latest px4fmu-v2_default (with EKF2 enabled) and it touched down, then flipped over.  

See [image]

See [Log File](http://logs.uaventure.com/view/2SMJjtC9KqnbdeYDSyzbyM) which contains the entire flight, which was simply a takeoff to 3 meters, then a auto-landing.  Actuators are stable for nice stable takeoff and hover, but last 5 seconds they freak out and tip over the drone. There's no RC input at this point, and wouldn't matter if there was because it was in AUTO_LANDING mode.  Distance sensor shows a nice smooth landing curve that returns to zero.

I'm curious how to debug this one, love some pointers so I can fix the code and try again...

lovettchris, 2017-01-13:
the only thing I can think of after studying these logs is that the negative value on altitude caused the set point position control to think it needed to accelerate to reach ground level, a couple feet stuck in the grass causing it to tip over.  If this is the case the landing code needs a major rewrite, it should not try to accelerate (unless you are worried about landing in water?).  And it should also have logic to detect that it is tipping over and to stop trying to accelerate in that case.

tops4u, 2017-01-14:
I had a similar case after landing in PosCtl Mode. Your GPS is getting worse after landing and PosX/Y show rather large deviations. 

I'm no specialist in this, but maybe Landing was not yet detected and it tried to maintain position which is 6+ meters away after touchdown.

LorenzMeier, 2017-01-14:
This should fix it: https://github.com/PX4/Firmware/pull/6327

lovettchris, 2017-01-27:
Yes, landing is much better now. Thanks.

closed 2017-01-27 by lovettchris

cimbar, 2018-08-30:
I've faced with the same situation at 1.8.0 version. I was tuning landing parameters and last few landings were really smooth, there wasn't any problem. Then, last try was terrible. After the vehicle touched the ground, it flipped exact one cycle and hit the ground again. I couldn't find any strange thing in the log. 
https://review.px4.io/plot_app?log=21562b84-1139-4c74-99f0-6c4c9c72b870

tops4u, 2018-08-31:
I'm no real Specialist, but noticed a few things that are strange in your log:
- Log speed it to low to really see details, maybe use faster logging for IMU in order to get more Data
- Any special Reason you use LPE?
- Hovering Throttle is rather high (almost 70%) maybe you should check your Setup, same goes for MIN Throttle. As your Min PWM is also already 1075 min Throttle would be 1300us / 25%
- You started **without** having a GPS Lock which was only available while already in Landing Mode
- Are you using external Position Estimation? Or why is the Vision Bit set in the LPE Bitmask?