PX4/PX4-Autopilot #6310: PX4 simulator kills all precesses which contain 'px4' in name

jdzerve, 2017-01-12:
There is a very appalling multiple instance detection built in px4 simulator.
It contains command "pgrep px4 && pkill px4"
This effectively kills all processes which happen to have 'px4' in their name.
simple solution would be to add anchors:
pkill '^px4$' so that it would match only 'px4' and not 'apx44'
Of course more sophisticated lock file based scheme would be much nicer.

dagar, 2017-01-12:
Thanks @jdzerve . Could you open a pull request that fixes it to your liking?
https://github.com/PX4/Firmware/blob/master/Tools/sitl_run.sh#L61
It needs to work on OSX and Linux.

jdzerve, 2017-01-13:
Created Pull request for this: https://github.com/PX4/Firmware/pull/6319

closed 2017-01-13 by jdzerve