PX4/PX4-Autopilot #6315: new feature request: Fixed-Wing style RTL

Antiheavy, 2017-01-13:
I'd like to propose a set of new features to PX4 to support a more appropriate Fixed-Wing style RTL (Return to Land).  Original post was here: http://discuss.px4.io/t/question-about-fixed-wing-rtl/2073/5

I'm still coming up to speed on PX4, but I have a lot of experience with small UAVs.  I don't have resources to help write code, but I can help work through the details.  I DO have resources to help flight test it as we try to get out every few days.

A potential Fixed-Wing style RTL solution might look something like the following.

1) Any time RTL is triggered the aircraft should go to a MAV_CMD_DO_LAND_START command/location in the mission (required to be in the mission in this case).  for example: pressing the [RTL] button in Qgc, auto RTL from low battery (BAT_CRIT_THR?), auto RTL from Loss of Comms (NAV_DLL_CH_T expires without regaining comms?), and maybe auto RTL in Loss of RC (NAV_RCL_LT expires?).

2) Other parameters associated with "Return to Land" should still work, although at the MAV_CMD_DO_LAND_START location instead of the "HOME" location. (e.g. RTL_RETURN_ALT, RTL_DESCEND_ALT, RTL_LAND_DELAY, RTL_MIN_DIST).
https://pixhawk.org/firmware/parameters#return_to_land1

3) Does Qgc currently place both the Comms Hold Waypoint and the Airfield Home Waypoint at the "HOME" point in Qgc? Am I correct to assume NAV_DLL_CH_T occurs while loitering above the "Home" point? Either way the fixed-wing aircraft should go to RTL as described above when NAV_DLL_CH_T expires without regaining communications.
https://pixhawk.org/firmware/parameters#data_link_loss1

4) I suppose folks with hybrid aircraft (e.g. quad planes, etc) might want the ability to choose between VTOL or Fixed-Wing RTL types. Maybe "RTL type" could be selected as a parameter?

5) If Fixed-Wing RTL type is selected there should be a mission validity check that requires a valid landing waypoint and valid MAV_CMD_DO_LAND_START point before accepting the upload (this could happen in PX4 and/or Qgc).  This is important because the HOME point will not be used for RTL.

Thank you for your consideration.

dagar, 2017-01-13:
1. With MAV_CMD_DO_LAND_START as the landing of RTL we should get all of this for free.
 2. The existing return to land has RTL_LAND_DELAY which causes the vehicle to optionally loiter over HOME before landing. How would this extend to land via MAV_CMD_DO_LAND_START? 
 3. These aren't widely used, and aren't exposed by the GUI. Maybe they should be?
 4. We should double check, but I believe the existing RTL behaviour for VTOL is optimal. Only transition before landing.
 5. Great idea!

**Next steps**
The key thing from my perspective is the importance (and implementation details) of #2. Originally I was thinking of having RTL with MAV_CMD_DO_LAND_START internally just be mission mode starting at the landing waypoint. 
We should probably have a new param to configure if RTL landing is at HOME or MAV_CMD_DO_LAND_START if available.
We also need to decide what happens if RTL_LAND_DELAY is set.

Antiheavy, 2017-01-13:
1.  ok

2. I think RTL_LAND_DELAY could be a useful function as long as the loiter was over the MAV_CMD_DO_LAND_START location instead of HOME.  I assume setting this to 0 would essentially cause it to be unnoticed?  If it is set it to >0 then the aircraft would hold altitude in the loiter about MAV_CMD_DO_LAND_START location for a period of time.  This might be useful if RTL_RETURN_ALT is high and RTL_DESCEND_ALT is much lower.  The aircraft will loiter-descend down to RTL_DECENT_ALT at the new MAV_CMD_DO_LAND_START location before breaking out to land and some might want the aircraft to achieve a stable loiter altitude before breaking out.  Ideally the aircraft would break out in a smart way as to most directly get on the landing glide slope after achieving stable RTL_DECENT_ALT, but I think that might be a follow-on problem to solve.

3. maybe Comms hold point and airfield home point aren't important.  For Loss of Comms it is nice to loiter over the HOME point before proceeding to the MAV_CMD_DO_LAND_START location.  For low battery and a manual RTL button press the aircraft should head directly to the MAV_CMD_DO_LAND_START location while climbing to the RTL_RETURN_ALT.

4. ok
5. ok

dagar, 2017-01-15:
One minor thing is that MAV_CMD_DO_LAND_START doesn't actually have a position, it's just a marker placed in the mission sequence. The position will be the following waypoint.

Right now I have a solution for RTL land initiating the mission starting at MAV_CMD_DO_LAND_START, however it switches the mode from RETURN -> MISSION, which might be abrupt/confusing. I'll think about other options.

The other piece is updating the mission feasibility checker to make sure a valid landing follows MAV_CMD_DO_LAND_START.

Antiheavy, 2017-01-17:
Sounds good.  I am available to discuss details if you like.  We can test whenever you have a test build ready.

dagar, 2018-01-17:
Done!