PX4/PX4-Autopilot #6330: Unable to land via landing command in MAVROS in GPS denied environment

myeabhishek, 2017-01-14:
I am working on offboarding using MAVROS. I have a node that issues setpoints to the drone and a separate node for issuing landing command.
In order to land the drone, I have to first kill the first node that issues the setpoints and then give landing command. In GPS-In environment this works fine but in case of GPS denied environment my landing command is rejected and drone goes into fail safe which lands it eventually.
I have set my fail safe to be loitering through QGC but even then it is showing the same behaviour.

Also, this particular started occurring since I last updated my firmware. Prior to that, everything was working fine.

Please help fix this.

LorenzMeier, 2017-01-20:
We cannot land without GPS - what are you trying to do exactly? Do you have another positioning source?

LorenzMeier, 2017-01-30:
Land is a normal operation command, failsafe is something entirely different. We will not add no GPS support for landing

closed 2017-01-30 by LorenzMeier