PX4/PX4-Autopilot #6332: rsync error

FantasyJXF, 2017-01-14:
I compile the Firmware on my **Windows 7**

```
git clone https://github.com/PX4/Firmware.git
cd Firmware
git submodule update --init --recursive
make px4fmu-v2_default
```

it runs out 
```
[  3%] Copying NuttX for px4fmu-v2 with nsh
[  4%] Copying NuttX for px4io-v2 with nsh
The source and destination cannot both be remote
rsync error: syntax or usage error (code 1) at /usr/src/rsync/rsync-3.0.8/main.c(1148) [receiver=3.0.8]
make[3]: *** [nuttx_copy_px4fmu-v2.stamp] Error 1
make[2]: *** [CMakeFiles/nuttx_copy_px4fmu-v2.dir/all] Error 2
make[2]: *** Waiting for unfilished jobs....
The source and destination cannot both be remote
rsync error: syntax or usage error (code 1) at /usr/src/rsync/rsync-3.0.8/main.c(1148) [receiver=3.0.8]
make[3]: *** [src/modules/px4iofirmware/nuttx_copy_px4fmu-v2.stamp] Error 1
make[2]: *** [src/modules/px4iofirmware/nuttx_copy_px4fmu-v2.dir/all] Error 2
make[1]: *** [all] Error 2
make[1]: *** [px4fmu-v2_default] Error 2
```

and I just cant find the file `nuttx_copy_px4fmu-v2.stamp`

Hoping for your answer ^_^

dagar, 2017-01-14:
Could you go into the build directory build_px4fmu-v2_default/ and run `make VERBOSE=1`?
That way it will print that paths used for rsync.

FantasyJXF, 2017-01-14:
It's 11:00PM in China ,I will try tomorrow ,I'm sorry and thank you

FantasyJXF, 2017-01-15:
[image]

Still don't know what is "rsyns" -_-

dagar, 2017-01-15:
I'm guessing we need to use relative paths with rsync to work around the cygwin path confusion.

dagar, 2017-01-15:
@FantasyJXF could you try the branch win_nuttx_rsync?

FantasyJXF, 2017-01-15:
Should I change the cmake/nuttx/px4_impl_nuttx.cmake to 

```
file(RELATIVE_PATH nuttx_cp_src ${PX4_BINARY_DIR} ${PX4_SOURCE_DIR}/NuttX)
	add_custom_command(OUTPUT nuttx_copy_${CONFIG}.stamp
		COMMAND ${MKDIR} -p ${nuttx_src}
		COMMAND rsync -a --delete --exclude=.git ${nuttx_cp_src}/ ${CONFIG}/NuttX/
		COMMAND ${TOUCH} nuttx_copy_${CONFIG}.stamp
		DEPENDS ${DEPENDS} ${nuttx_patches} ${nuttx_all_files}
		COMMENT "Copying NuttX for ${CONFIG} with ${config_nuttx_config}"
		WORKING_DIRECTORY ${PX4_BINARY_DIR}
		)
	add_custom_target(nuttx_copy_${CONFIG} DEPENDS nuttx_copy_${CONFIG}.stamp)
```

And try the reinstall the PX4 Toolchain then repeat the steps `git clone https://github.com/PX4/Firmware.git`  `git submodule update --init --recursive`

What you said about "bracn win_nuttx_rsync" , I have no idea about it .

I tried 'git branch -a ' and didn't see THE branch

dagar, 2017-01-15:
If you fetch (git fetch --all) you should see it. Otherwise you can manually change the 2 lines to test. If this works it'll go into master shortly.

FantasyJXF, 2017-01-16:
Still not OK.  I reinstall the px4 toolchain, and conducted the following steps:

1 ) git clone https://github.com/PX4/Firmware.git

2) WIP: cmake nuttx rsync copy only use relative paths #6342

3 ) git submodule update --init --recursive
     make px4fmu-v2_default

it runs into a different problem

[image]

4 ) make clean 
git checkout  win_nuttx_rsync
make px4fmu-v2_default

it got the same wrong message as the above picture?

Shouldn't I do the second step?

dagar, 2017-01-16:
It looks like you have gcc (mingw) 4.6.2. Can you update to at least 4.8 or 4.9? For master the build system is currently using 5.4.

FantasyJXF, 2017-01-16:
Usually the default gcc is gcc-arm-none-eabi 4.7.4, I use the gcc5.4.3 to covered same folders in the px4/toolchain .

On master branch , the `make px4fmu-v2_default` get another error message

[image]
 
On  bracn ch win_nuttx_rsync ,it also failed

[image]

It seems to be the code

dagar, 2017-01-16:
I would guess your submodules are now out of sync.
Try `make distclean` and then build again.

FantasyJXF, 2017-01-16:
`make distclean` seems not to be a valid command
[image]

dagar, 2017-01-16:
It definitely is. We must be using some git submodule argument that isn't available in an older version, or yet another thing subtly broken on windows.

Can you try.

    git clean -ff -x -d
    git submodule sync --recursive
    git submodule update --init --recursive --force

We just want to make sure absolutely everything is clean and updated, the just building again with a more recent GCC.

FantasyJXF, 2017-01-16:
good news ,before taking your advice ,I run the `make px4fmu-v2_default` again. It successes.

But you might want to see the following pictures 

[image]
[image]
[image]

my gcc is 5.4.3,and the third picture showed the 4.6.2 , it confused me a lot .

but  anyway, finally I got 
[image]

I'm going to try add a uORB topic and make it again, let's see what will happen.

FantasyJXF, 2017-01-16:
I really realy realy appreciate your help these days, all the time I work alone and there are so so many problems . I'm gong to figure it out what is the PIXHAWK

closed 2017-01-16 by LorenzMeier

lairay, 2017-10-15:
hello @FantasyJXF  :
   I have a totally the same problem with you about the "usr/src/rsync/rsync-3.0.8/main.c(1148)" error.
I read this page and try,but I still can not fix my problem. It stuck in built target nuttx_builtins.

Built target git_nuttx
[  2%] Built target uorb_headers
[  2%] Built target nuttx_builtins
[  2%] Copying NuttX\apps to D;D:\MyCopter\pixhawk_toolchain\toolchain\msys\1.0\git\Firmware\build\px4fmu-v2_default\NuttX
[  3%] Copying NuttX\apps to D;D:\MyCopter\pixhawk_toolchain\toolchain\msys\1.0\git\Firmware\build\px4io-v2_default\NuttX
[  2%] Copying NuttX\nuttx to D;D:\MyCopter\pixhawk_toolchain\toolchain\msys\1.0\git\Firmware\build\px4fmu-v2_default\NuttX
The source and destination cannot both be remote.
rsync error: syntax or usage error (code 1) at /usr/src/rsync/rsync-3.0.8/main.c(1148) [Receiver=3.0.8]
make[3]: *** [NuttX/apps_copy.stamp] Error 1
make[3]: *** Waiting for unfinished jobs....
The source and destination cannot both be remote.
rsync error: syntax or usage error (code 1) at /usr/src/rsync/rsync-3.0.8/main.c(1148) [Receiver=3.0.8]
make[3]: *** [NuttX/nuttx_copy.stamp] Error 1
make[2]: *** [NuttX/CMakeFiles/nuttx_configure.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  3%] Copying NuttX\nuttx to D;D:\MyCopter\pixhawk_toolchain\toolchain\msys\1.0\git\Firmware\build\px4io-v2_default\NuttX
The source and destination cannot both be remote.
rsync error: syntax or usage error (code 1) at /usr/src/rsync/rsync-3.0.8/main.c(1148) [Receiver=3.0.8]
make[7]: *** [NuttX/apps_copy.stamp] Error 1
make[7]: *** Waiting for unfinished jobs....
The source and destination cannot both be remote.
rsync error: syntax or usage error (code 1) at /usr/src/rsync/rsync-3.0.8/main.c(1148) [Receiver=3.0.8]
make[7]: *** [NuttX/nuttx_copy.stamp] Error 1
make[6]: *** [NuttX/CMakeFiles/nuttx_configure.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [px4io-v2_default] Error 2
make[3]: *** [../px4io-v2_default/nuttx_px4io-v2_default.elf] Error 2
make[2]: *** [src/drivers/px4io/CMakeFiles/copy_px4io_bin.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [px4fmu-v2_default] Error 2

Could you tell me how you solved your problem?

Seems that you are living in China, I can speak in Chinese. 
You can contact me with my e-mail : larryray618@gmail.com

FantasyJXF, 2017-10-15:
@lairay  I think this problem has been solved by the official, I haven't met this question for a lonf time, you don;t need to modifiy anything

lairay, 2017-10-15:
But I still can't success in run "make px4fmu-v2_default". I can't  find "usr/src/rsync"this folder and so do the main code /rsync-3.0.8/main.c(1148)

FantasyJXF, 2017-10-15:
Make sure you have installed all the dependecies on the dev guide.

And maybe you can change a computer or use a virtual machine to do this, btw, I use Ubuntu now.

lairay, 2017-10-15:
Okay. thank you. I will try that.

dagar, 2017-10-15:
This might be a new error, rsync on windows is a pain because of the confusion between cygwin and native paths. The PX4 NuttX build was recently overhauled, but unfortunately we don't have any Windows continuous integration, so it tends to break.

Please make sure all your windows dependencies are up to date, you're on the latest PX4 master, then if the build still fails open a new issue and tag me. Post the full build text, then try again with `make px4fmu-v2_default VERBOSE=1` and post that full output as well.

dagar, 2017-11-30:
@lairay could you please try this PR? https://github.com/PX4/Firmware/pull/8397

PR #6342 (2017-01-16): WIP: cmake nuttx rsync copy only use relative paths
 - this is to work around confusion between cygwin and windows native
   paths
 - closes #6332 

Waiting for testing on windows (#6332).
commit 73b31c03 (2017-01-16): cmake nuttx rsync copy only use relative paths

 - this is to work around confusion between cygwin and windows native
   paths
 - closes #6332