PX4/PX4-Autopilot #6334: Qt runs out `Built target px4` NOT `Built target px4fmu-v2_default`

FantasyJXF, 2017-01-14:
On my Ubuntu 14.04 LTS  Terminal

I type `make px4fmu-v2_default` and get `Built target px4fmu-v2_default`

then I use the following commands to get some files for Qt

```
cd ~/src/Firmware
mkdir ../Firmware-build
cd ../Firmware-build
cmake ../Firmware -G "CodeBlocks - Unix Makefiles"
```

Qt runs out [100%] Built target px4 but not Built target px4fmu-v2_default? 
I check the CMakeLists.txt and change LINE 143 to set(CONFIG "nuttx_px4fmu-v2_default" CACHE STRING "desired configuration") but didn't work

Hope to talk with you soon.

And I run an PX4 wiki China 

https://fantasyjxf.gitbooks.io/px4-wiki/content/index.html

dagar, 2017-01-14:
Try deleting the build directory and running cmake again with -DCONFIG

    cmake ../Firmware -G "CodeBlocks - Unix Makefiles" -DCONFIG=nuttx_px4fmu-v2_default

FantasyJXF, 2017-01-16:
it works.Thanks. Next I'm going to solve the "rsync error" on Windows 7 , there are many people encounter  with the same problem

dagar, 2017-01-16:
Glad to hear!

closed 2017-01-16 by dagar