PX4/PX4-Autopilot #6337: SITL Drone rolls to the right

nicolaerosia, 2017-01-14:
Hello,

I'm trying to manually control the drone in SITL Gazebo by using an [RC joystick](https://hobbyking.com/en_us/hobbyking-6ch-rc-flight-simulator-system-mode-2.html) connected to QGroundControl.
I'm instructing the drone to take off (`commander takeoff`) and after that I'm switching the drone to MANUAL mode using the QGroundControl menu.
As soon as I do that, the drone starts rolling to the right. If I move the roll joystick to maximum left, the roll almost disappears.

Complete workflow:
1. start PX4
`make posix_sitl_default gazebo_iris`
2. Start QGroundControl
3. Calibrate Joystick
4. `pxh> commander takeoff`
5. Switch to MANUAL from QGroundControl

I believe I have tested this in the past and it was working. Am I missing something obvious?
Is anyone else testing SITL using a USB Joystick?

dagar, 2017-01-16:
@nicolaerosia can you try setting COM_RC_IN_MODE to "Joystick/No RC Checks"?

nicolaerosia, 2017-01-16:
@dagar That's the default value (actually set by startup script), and it has the roll problem.
I ended up testing also with COM_RC_IN_MODE to "Virtual RC by Joystick", and as you can see in the second image, the Joystick looks calibrated, but the values reported by PX4 for Radio's second channel have an offset.

dagar, 2017-01-16:
With COM_RC_IN_MODE 1 (Joystick/No RC Checks) the actual calibration is on the QGC side. For COM_RC_IN_MODE 2 (Virtual RC by Joystick) you calibrate through QGC, but the calibration is applied by the firmware.

It's unlikely this is a problem in the firmware.

For a sanity check I'd try uninstalling QGroundControl and delete the settings directory, then install the latest stable (http://qgroundcontrol.com/downloads/).
Then on the firmware side to be paranoid you could clean (make distclean) and rebuild. This ensures all the params are reset.

Within QGC keep the default COM_RC_IN_MODE, calibrate joystick only, and try flying again.

nicolaerosia, 2017-01-16:
Thank you for the suggestions, I cleaned up my workspace:
```
rm -rf /home/user/.cache/QGCMapCache300 /home/user/.config/QGroundControl.org /home/user/.local/share/QGroundControl.org
```
tried QGroundControl stable and daily (cleaning in between) and now it cannot see my Joystick.

Any pointers on how to debug this?

dagar, 2017-01-16:
No, but clearly we've got some work to do with QGC joystick handling. Can you open an issue over there to continue the discussion?

Now that you've completely cleaned everything you could also retry the previous stable. https://github.com/mavlink/qgroundcontrol/releases

closed 2017-01-16 by dagar

nicolaerosia, 2017-01-16:
Thanks! It was a QGroundControl bug, it works on v3.0.2