PX4/PX4-Autopilot #6345: EKF2 log replay problem

meteor0530, 2017-01-16:
Hi
My system is MAC OS. I run replay tool according to the developer guide, and I dont know how to continue from this step.

[image]
I need some help about the replay tool.
Thanks.

tumbili, 2017-01-16:
@meteor0530 One of the upper line in the picture above tells you where your replay log will be saved to. You should be able to open that log and view the results, e.g. using FlightPlot.
https://dev.px4.io/flight_log_analysis.html

You might also be interested in this
https://dev.px4.io/tuning_the_ecl_ekf.html

Btw, assuming that the replay tool is working correctly, the GPS fusion timeout messages you are getting mean that the estimator is constantly resetting it's position. You will have to find out what's causing it (e.g. bad IMU calibration, ...)

meteor0530, 2017-01-16:
@tumbili Thank you, but I can't find replay_replayed.px4log in that file. (actually, there is no file named 'rootfs' below posix)

tumbili, 2017-01-16:
@meteor0530 Ah, I think that location might have changed on master and the message should be updated. I'll change the header of this issue then. Try to look under build_posix.../tmp/
In case you don't find it try to search for the file in the build directory.

tumbili, 2017-01-16:
This is not correct anymore:
https://github.com/PX4/Firmware/blob/master/src/modules/ekf2_replay/ekf2_replay_main.cpp#L836

meteor0530, 2017-01-16:
@tumbili Thank you for the help, I have found the replay log. 
After i change the params in replay_param.txt, how to update the replay log?

tumbili, 2017-01-16:
@meteor0530 After you have changed the parameters in the file you just need to re-run the replay which will use the updated parameters. It will also write those parameters to the replay log file.

meteor0530, 2017-01-16:
@tumbili In order to get the log, do i need to fly my vehicle outdoor and switch to posctrl or just disarm it  on the ground?
Once i flied my vehicle outdoor and switch to posctrl, it could not hover and fly away, is it caused by the wrong params of ekf2?

dagar, 2017-04-12:
@meteor0530 is this resolved? Please reopen the issue if it's still a problem.

closed 2017-04-12 by dagar