PX4/PX4-Autopilot #6347: Flow out-of-focus takeoff support

LorenzMeier, 2017-01-16:
Current discussion state:

- Report flow as valid or special mode if landed (if flow data is coming in)
- Report sonar / altimeter as zero if landed (if altimeter data is coming in)
- Once non-landed time-out after 2-3 seconds if either flow or sonar / altimeter do not become valid
- Ensure position controller goes into failsafe if timing out
- Takeoff should takeoff action, not user-driven altitude (so the system clears the minimum altitude quickly)

LorenzMeier, 2017-01-16:
@ChristophTobler @priseborough @tumbili

ChristophTobler, 2017-01-16:
About flow failsafe: 
* In the simulation it does switch to ```WARN  [commander] failsafe mode on``` after not getting valid measurements (fly outside of asphalt) after a few seconds. But then it just switches to altitude mode (obviously it can not return when it does not have a position information). I guess it should handle that case by just landing?
* Do what is mentioned above. Then there seems to be a bug when returning to the asphalt manually. It does not always switch back to ```WARN  [commander] failsafe mode off```

LorenzMeier, 2017-01-16:
Yep, we should add auto-land as an option instead of falling back to manual and we should find and fix the issue which is not making it return.

mhkabir, 2017-01-16:
@jgoppert You will be interested in this. We should also consolidate the hack in LPE which we currently use to do the out-of-focus takeoffs.

jgoppert, 2017-01-17:
Any reason not to implement a fake landing measurement like LPE does. This gives agl = 0, vx=0, vy=0 when landed. During takeoff the measurement goes away and the vehicle is in the air. Then the covariance matrix for x, y is growing. If it gets too large it publishes that position is invalid. So you get a timeout that is adaptive based on the measurements that you have. I think this is the best option.

jgoppert, 2017-01-17:
@priseborough That change looks reasonable to me. The only other thing you might want to do is add a measurement of agl = 0 to keep the terrain state and altitude state the same.

priseborough, 2017-01-17:
What I did with the APM implementation was add a parameter which was the expected on-ground range measurement. In the absence of range data it would use the expected on-ground value if not in-air. This allowed for the variation in landing gear height between copters.

jgoppert, 2017-04-28:
I think both estimators have working solutions for this now.

closed 2017-04-28 by jgoppert

mhkabir, 2017-04-28:
I'm not sure if this is solved. @ChristophTobler?

ChristophTobler, 2017-04-29:
Yes, this is solved. Thanks