PX4/PX4-Autopilot #6349: Bebop: Auto-start app

LorenzMeier, 2017-01-16:
@eyeam3

mhkabir, 2017-02-05:
On Bebop 2, modify the init script /etc/init.d/rcS_mode_default. Comment the following line:
```
DragonStarter.sh -out2null &
```
Replace it with:
```
px4 /home/root/px4.config
```

How to : 
Enable adb server by pressing the power button 4 times.
Connect to adb server as described before:
```
adb connect 192.168.42.1:9050
```
Re-mount the system partition as writeable:
```
adb shell mount -o remount,rw /
```
In order to avoid editing the file manually, you can use this one : 
<script src="https://gist.github.com/mhkabir/b0433f0651f006e3c7ac4e1cbd83f1e8.js"></script>
Save the original one and push this one to the Bebop
```
adb shell cp /etc/init.d/rcS_mode_default /etc/init.d/rcS_mode_default_backup
adb push rcS_mode_default /etc/init.d/
```
Sync and reboot:
```
adb shell sync
adb shell reboot
```

mhkabir, 2017-02-05:
I would put it into Devguide myself, but I still can't edit it. Seems to be a permissions issue, because everyone else seemingly can.

mhkabir, 2017-02-05:
Done

closed 2017-02-05 by mhkabir