PX4/PX4-Autopilot #6355: Transition between manual and auto flight

LorenzMeier, 2017-01-16:
@MaEtUgR With the position controller changes the transition between manual and auto is a bit abrupt. I think we should set the hover throttle to the manual throttle, constrained to +- 10% around the configured hover throttle.

Examples:

 * Case 1: manual at 45%, Hover param at 50% -> new auto throttle set to 45% on switch

 * Case 2: manual at 25%, Hover param at 50% -> new auto throttle set to 40% on switch

Right now its too abrupt and scary. Reported here: http://review.px4.io/plot_app?log=e159e728-6a7c-4d8d-a975-4da6b6a67f88

MaEtUgR, 2017-01-17:
Definitely, there is a bug in the transition that always generates a throttle spike away from the ground.
I think it's not the vertical thrust integral but maybe something about the velocity control smooth transition calculation.
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1741

MaEtUgR, 2017-01-17:
I also found the problem yesterday in our flight log here (it's the transition at about 2/3 flight time):
http://logs.uaventure.com/view/Rgiz7unTzVztqjkhaU9HSZ

lovettchris, 2017-01-18:
When I see logic in mc_pos_control that resets integrals I get worried about disjoint control behavior.  Should those be ramps instead of resets?  For example: 

              /* reset integrals if needed */
             _thrust_int(2) = 0.0f;

Does the integral really need to start over?  I would find that code easier to read if we had a PID class to encapsulate the PID algorithm.  Then perhaps we could localize the complexity of making a smoother PID algorithm in that class rather than sprinkling it over 2420 lines of mc_pos_control code...

Stifael, 2017-01-27:
@lovettchris: thats actually the goal, but its not something that can be done easily without making sure that nothing breaks. I think once we have tests that cover most of the mc_pos_control module, we then can start splitting the code into different classes etc.

LorenzMeier, 2017-01-27:
@lovettchris Its unlikely you can handle this with a generic PID class any better. The reason is that the interactions in the attitude controller are trigonometric and most of them stem from multi rotors being underactuated.

What we can improve is encapsulation to make it more readable. But the complexity will likely continue to persist and limits and boundary conditions will likely be sprinkled across the codebase.

Mostly because a lot of them are use-case specific *and* situation specific.

LorenzMeier, 2017-04-30:
I think we fixed this.

closed 2017-04-30 by LorenzMeier

lovettchris, 2017-04-30:
yes works much better now, thanks.

MaEtUgR, 2017-05-09:
I think https://github.com/PX4/Firmware/commit/34080be68bb9fee0b67c1e868cb024515bfbd486 was the fix for the big spike when switching mode.