PX4/PX4-Autopilot #6358: SF1XX Driver not activated on px4fmu-v2

mzahana, 2017-01-17:
Hi,
I was trying to use sf11/c sensor as an altimeter with Pixhawk 1 (px4fmu-v2). However, it did not work as it is not added to the corresponding cmake file [here](https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake#L30)

Is there a reason not having this available?

mzahana, 2017-01-17:
> We don't have enough flash space to fit it into. You'll need to add it and
> compile.

I did that, and it compiled fine. No space issue on Pixhawk 1. What is the
space issue you are talking about then?

dagar, 2017-04-11:
This has been readded to px4fmu-v2_default.

closed 2017-04-11 by dagar

pythonmobile, 2017-04-24:
@dagar Can you please look at this:
https://github.com/PX4/Firmware/issues/7120

@mzahana I did hit the space issue that @mhkabir talked about?

hunterakins, 2017-07-26:
Hi,
I am also trying to add some support for SF1XX using px4fmu-v2_default

I am looking in /drivers/boards/px4fmu-v2/init.c and I can see that there is no i2c support

I have added /drivers/sf1xx to the make file for nuttx-px4fmu-v2_default.cmake and that doesn't help.

I think a good start would simply be to enable the i2c debugger, the nuttx application, but not sure where to start. 

Any help would be great.

Thanks,
Hunter

dagar, 2017-07-28:
@hunterakins you need to add it to this file. https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake

You need this line https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v4_default.cmake#L51

Please use discuss (http://discuss.px4.io/) for these kinds of questions or open new github issues (for bugs).

hunterakins, 2017-07-28:
I have that line in my cmake. 
When I open up nsh I look in /dev and I don't see /dev/sf1xx there, which the driver uses, that's my error. 
I will open an issue

dagar, 2017-07-28:
My guess would be the device is only created if the driver is started successfully. Did you set SENS_EN_SF1XX?

hunterakins, 2017-07-28:
Yes I set SENS_EN_SF1XX to 4

hunterakins, 2017-07-28:
As per 
https://dev.px4.io/en/sensor/sf1xx_lidar_setup.html

dagar, 2017-07-28:
Is it a serial console or nsh over mavlink? If the former than please share anything relevant from bootup. Did it try to start sf1xx and fail, etc.
I'm not familiar with the implementation of this driver, but it probably stops if the device wasn't found. You might also want to double check your wiring. 

Is this one i2c or serial? Which board are you using and where is the lightware physically connected?

hunterakins, 2017-07-28:
It might be helpful to know that I am using a Pixhawk2 board, using the I2C2 connection. I think the driver may not work with the newer pixhawk.

hunterakins, 2017-07-28:
Connected over serial console. 
It does try and fail: here's the message from bootup
ERROR [sf1xx] driver start failed
[image]

dagar, 2017-07-28:
Okay that makes sense. I took a quick look at the sf1xx driver and it's only trying what the pixhawk and other boards considered the external bus. 

The pixhawk 2.1 cube took what the pixhawk considered internal i2c and made it availabile as I2C2.
Try hacking it to use PX4_I2C_BUS_ONBOARD here. https://github.com/PX4/Firmware/blob/master/src/drivers/sf1xx/sf1xx.cpp#L80
If that works we can find a more permanent fix.
@davids5 FYI

hunterakins, 2017-07-28:
I tried that, no luck. 
Thanks for the help by the way, I've been struggling with this for a while.

dagar, 2017-07-28:
Ping me on slack if you'd like to try a few more things. http://slack.px4.io/

mhkabir, 2017-07-29:
@hunterakins OK, I know what's going on. I hacked around this on my setup, but what is happening is that the driver is looking for the lidar on PX4_I2C_BUS_EXPANSION, but I2C2 on Pixhawk 2.1 is PX4_I2C_BUS_INTERNAL. With the more fine-grained board config we now have from @davids5, I think it should be possible to fix this.

hunterakins, 2017-07-31:
@mhkabir Okay thanks, figured out the problem, I forgot to supply external power...sort of figured the pixhawk  5V on I2C would make it work but the SF11C requires power to the microusb input as well. Stupid of me. At any rate, I did confirm that it only works when making the change @dagar suggested: changing PX4_I2C_BUS_EXPANSION to PX4_I2C_BUS_ONBOARD. Note: setting it to PX4_I2C_BUS_INTERNAL does not compile, it has to be ONBOARD for me.

mhkabir, 2017-08-01:
You shouldn't need to supply external power. If your wiring and/or hardware is OK, the Pixhawk can easily power it.

dagar, 2017-08-01:
I think it's that USB power isn't sufficient.

bozkurthan, 2019-03-19:
Hello, I don't want to create new issue but I have still problem with SF11/C laser range finder. 
I'm working with Cube ( Pixhawk 2.1 ) and I tried to connect SF11/C  like described here [https://docs.px4.io/en/sensor/sfxx_lidar.html](url) and wired correctly to Telemetry 2. I connected Cube to PC via micro USB cable. When I don't change `SENS_SF0X_CFG` parameter to Telemetry 2. I monitored SF11/C's values from Mavlink console. ( I used `cat /dev/ttyS2` command to perform this).
So, I don't think I have power problem. Also, I tried the SF11/C on I2C input and no change.
I'm still struggling with this issue. What is the problem? Thanks in advance for any help.

bozkurthan, 2019-03-19:
Screenshots:

Wired connections:

[image]

[image]

Setup without parameter setting via Telemetry 1:
[image]

Setup with parameter via Telemetry 2:

[image]

[image]

[image]

[image]

[image]

[image]

@dagar and @mhkabir

hamishwillee, 2019-03-19:
@bozkurthan Please open a new issue. As a matter of principle no one should look at this.

bozkurthan, 2019-03-19:
Ah ok. I will create new issue thanks for information.