PX4/PX4-Autopilot #6364: EKF2 Does Not Work with MOCAP

mzahana, 2017-01-17:
Hi,

Is there a way to use EKF2 with Mocap? I could not get it working just by sending MOCA_ATT_POS message.

Also it does not make sense, that I use e.g. LPE to do some test indoors with MOCAP, then switch to EKF2 to do the same tests in outdoors.

It would make the comparison more realistic if EKF2 can work in both outdoors (with GPS), and indoors (faked GPS using MOCAP).

So, is there a way to make that possible with the current EKF2 implementation?

Thanks.

mzahana, 2017-12-14:
This issue has been there for a year now! Is there any update regarding fusing MOCAP into EKF2?

I saw some work on this in this issue https://github.com/PX4/ecl/issues/271

mzahana, 2018-01-03:
#8520

jannsta1, 2018-07-22:
Hi @mzahana, just wondered if you're now using EKF2 with mocap? I noticed that in firmware 1.8.0 lpe is no longer supported so probably time to make the switch if EKF2 working!

mzahana, 2018-07-23:
@jannsta1 Yes. I am fusin mocap data into EKF2.

jannsta1, 2018-07-30:
Thanks @mzahana , my initial trial didn't go especially well, perhaps because I didn't reduce EKF2_EV_DELAY yet. I was wondering if you could post a log with your setup for comparison? 

Also, could you confirm that you are sending the mocap data via the ```vision_pose_estimate``` topic? The [user guide](https://dev.px4.io/en/tutorials/motion-capture-vicon-optitrack.html) suggests ```mocap_pose_estimate``` but I think this might be legacy?

@hamishwillee (think you are working on the user guide?) This and linked issues could be a basis for updating the userguide to include parameters required for mocap with EKF2?

hamishwillee, 2018-07-30:
@jannsta1 I work on the PX guides, but these are just one of my priorities. So I am going to encourage you to do as much of the required work as possible - ie if you can get a setup working, I can help get everything documented.

I do know that a lot of work has gone into EKF2 and it does now support fusing optical flow - and is the recommended approach.

The docs we have have on EKF2 are mostly in the DEV guide (my understanding is that optical flow and mocap are the same thing?). I think these two are relevant:
* http://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html#optical-flow 
* http://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html#external-vision-system

Other than that the docs are all LPE-centric (drprecated). What we have is:
* [Flying with Motion Capture (VICON, Optitrack) > EKF2](http://dev.px4.io/en/tutorials/motion-capture-vicon-optitrack.html#ekf2)

  > The ROS topic for motion cap mocap_pose_estimate for mocap systems and vision_pose_estimate for vision. Check mavros_extras for further info.
* [Using Vision or Motion Capture Systems](http://dev.px4.io/en/ros/external_position_estimation.html)

Does any of that help?

jannsta1, 2018-07-31:
OK, I managed to get EKF2 working now - had previously blacklisted vision_pose_estimate in mavros 🙈 . Will experiment a bit with the settings and can post notes up once I know what's going on a bit better.

mzahana, 2018-08-04:
I did a test again and works fine with EKF2. I use Optitrack system and [vrpn_client_ros](http://wiki.ros.org/vrpn_client_ros) package.
Steps (briefly):
* Align the drone with the x-axis of mocap frame
* define rigid body
* configure mocap software to publish vrpn data
* configure mocap softwre to consider Z as the Up axis (default is Y in Optitrack)
* use `vrpn_client_ros` to get mocap data into ROS
* relay mocap pose topic to `/mavros/vision_pose/pose` using `topic_tools` package
* Adjust `EKF_AID_MASK` to fuse vision position and yaw
* adjust height parameter `EKF2_HGT_MODE` to use vision
* reduce `EKF2_EV_DELAY` if you are getting mocap pose at high rate. I reduced it to `50ms`. I don't know how to get accurate estimate of that parameter, though.
* reboot FCU for the parameters to take effect
* Make sure that `/mavros/local_pose/pose` are very close to the mocap pose topic

I guess that is all what I did to get it working .

I hope this helps.

LorenzMeier, 2018-08-04:
@mzahana A pull request against the PX4 user guide detailing these steps would be very much appreciated. @hamishwillee Could you keep track of this topic? Thanks!

hamishwillee, 2018-08-05:
@mzahana Do you think you will be able to create a PR in the user guide? I can help edit afterwards.

The main places that need updating are listed above in:  https://github.com/PX4/Firmware/issues/6364#issuecomment-409044467

Specifically these two:
* [Flying with Motion Capture (VICON, Optitrack) > EKF2](http://dev.px4.io/en/tutorials/motion-capture-vicon-optitrack.html#ekf2)
* [Using Vision or Motion Capture Systems](http://dev.px4.io/en/ros/external_position_estimation.html)

IT might even be worth merging those docs if possible?

Looking at your list above looks like good set of steps, but for instructions you'd want to add "how" to each point.
So, for example, the sort of info I might add (guessed) shown in italics:
1. Align the drone with the x-axis of mocap frame _as shown in the diagram_.
1. define rigid body _using parameter X_
1. configure mocap software to publish vrpn data _(by setting x, y, z)_
...

LorenzMeier, 2018-08-06:
Related issue: This user should also upgrade to EKF2: https://github.com/PX4/Firmware/issues/10169

beaucolley, 2018-08-06:
We have also just been able to implement Off-board mode with a VICON system running the EKF2 branch. Following steps as detailed above by mzahana.

https://review.px4.io/plot_app?log=aa944e78-073f-49fd-81ea-734d19aa0216

mzahana, 2018-08-06:
@LorenzMeier @hamishwillee , I can start a PR against the docs some time around the coming weekend.

mzahana, 2018-08-06:
@hamishwillee The steps I followed are ROS based. So, it probably makes sense to keep it under the ROS tutorials, specifically (http://dev.px4.io/en/ros/external_position_estimation.html). However, that page is still considering LPE. Since LPE is not supported anymore, I can replace it with instructions for EKF2, while keeping some of the general information on that page, e.g. reference frames. Let me know if that makes sense.

TSC21, 2018-08-06:
@mzahana not being supported doesn't mean it has to be dropped or replaced. I do think it should be extended besides LPE, not replace it, aa LPE is still supported (maintained and supported are different in meaning).

mzahana, 2018-08-06:
@TSC21 no worries.

mzahana, 2018-08-06:
@hamishwillee A PR is created https://github.com/PX4/Devguide/pull/573

beaucolley, 2018-08-06:
FYI I'm quite sure that there is a major bug in LPE causing the velocity estimate to be influenced by the MOCAP position. We experienced non zero starting velocities when the initializing away from the origin and velocity estimates that are a linear factor of the position. All were fixed when we switched to EKF2

hamishwillee, 2018-08-06:
@beaucolley Is there a bug report on that? If so then @mzahana Could add a note in docs of reported issues with LPE and link to the bug report.

beaucolley, 2018-08-07:
I haven't made a bug report, I created an [issue](https://github.com/PX4/Firmware/issues/10169#issuecomment-410600997) with some log files explaining the problem but switching to EKF2 fixed it so I closed it.

revantad, 2018-10-17:
@mzahana Hey, I am facing a similar issue and I see you have fixed it. What I am currently doing is publishing motion capture data to `/mavros/vision_pose/pose` and this data is visible on QGC in the correct format.
The parameters `EKF_AID_MASK`, `EKF2_HGT_MODE` and `EKF2_EV_DELAY` are set to the values you mentioned in your comment but when the `/mavros/local_position/pose` is echoed only the `Z` and `Orientation` values match but not the `X` and `Y` to the motion capture values.
If you could let me know what I might be missing, it would be great.

mzahana, 2018-10-17:
@revantad How does the X/Y values look like? Are they flipped?
Also, try to tune the `EKF2_EV_DELAY`; see [this comment](https://github.com/PX4/Devguide/pull/573#discussion_r209867749)

mzahana, 2018-10-17:
@revantad Try to tune the `EKF2_EV_DELAY`. Probably you need to decrease its value.

revantad, 2018-10-17:
@mzahana Thanks for the help. Will try it and update the thread.

shortcut21, 2018-12-14:
I have a similar problem. My motion capture device produces the position of my drone and it publishes it to the `/mavros/mocap/pose`  but when I use LPE it's not really accurate the follow of the setpoints I pass in offboard mode. On the other side when I use EKF without my motion capture my drone takes off and stays stable but floats(drifts away).  When EKF is enables and my drone is connected it doesn't get the /mocap/pose as a position. How can I combine my motion capture and EKF keeping in mind it is not vision pose?  
In other words, how to make my pixhawk get the pose from the motion capture and using EKF?
@beaucolley @revantad @mzahana Any ideas? 
Thanks! :)

hamishwillee, 2018-12-16:
@mzahana Can this be closed? To my mind it was addressed by https://github.com/PX4/Devguide/pull/573 and will be further tidied by https://github.com/PX4/Devguide/pull/681

closed 2018-12-17 by mzahana

vipin3690, 2022-10-22:
@mzahana I have a similar issue . My mavros/local_position/pose gives X and  Y values flipped .The altitude hold mode is working but position hold is working. The details of my issue are explained at #1792. I am also confused about what kind of modifications that should be done in the px4_config.yaml file.

hamishwillee, 2022-10-25:
@vipin3690 Please don't necrobump old dead issues. Ask on discord or discuss.