PX4/PX4-Autopilot #6367: Setpoints go stale for 15secs after back transition

sanderux, 2017-01-17:
[image]

dagar, 2017-01-17:
Full log?
Is 15 seconds your back transition timeout?

sanderux, 2017-01-17:
The problem is narrowing down to likely be this: https://github.com/PX4/Firmware/pull/5980
The vehicle lost a lot of altitude during backtrans due to difference in level quad/plane
[image]

Log: http://logs.uaventure.com/view/B5wvQbkXPAmDKpZJ49qGei

dagar, 2017-01-23:
The better solution is update the VTOL architecture to use fw_pos_ctrl to maintain altitude during back transition. With what we currently have you should try changing the pitch setpoint during back transition to be FW_PSP_OFF and don't reset the fw_att integrators at transition.

closed 2017-03-05 by sanderux