PX4/PX4-Autopilot #6369: Snapdragon Eagle: Sensor value spikes after calibration

LorenzMeier, 2017-01-17:
@wangxdflight @mcharleb @jywilson @bharath374 We do see sensor value spikes on Snapdragon flight after calibrating the gyro / accel. Could you please help us to reproduce this on your end?

@Stifael and @ChristophTobler found this today.

Getting the Snapdragon support streamlined really depends on driving it through the full use-case. To reproduce this, load master (using the latest Intrinsyc release) onto the board, reset the parameters (`param reset`) and calibrate using QGroundControl. Then run QGroundControl, go to the menu bar of your OS, select Widgets -> Analyze and look at the highres IMU. The code pieces are in the driver framework wrappers:
https://github.com/PX4/Firmware/tree/master/src/platforms/posix/drivers

In DriverFramework:
https://github.com/PX4/DriverFramework

And in the sensors app:
https://github.com/PX4/Firmware/tree/master/src/modules/sensors

LorenzMeier, 2017-01-17:
Preliminary investigation hints at the MPU9250 driver.

Stifael, 2017-01-23:
Current state:
I think something is just wrong with the MPU9250 driver. After having had the same three error on three different snapdragons (spikes), I had a successful setup with two other snapdragons. 
Today I also installed again the newest linux image on two of the snapdraon together with the newest dsp firmware from Intrinsyc. One of the snappies seems now to works without spikes. The other snappy does not show spikes anymore, but the gyro around the z-axis is sending just noise. Motion around the z-axis has no effect at all. Being not able to reproduce consistently the same error, I think something is most likely wrong with the MPU9250 driver.

wangxdflight, 2017-01-24:
I am working on it,,,trying to reproduce it first.

wangxdflight, 2017-01-24:
@Stifael Are you seeing this issue still on the master tip? I recently submitted 2 PRs, which fix some issues in shared memory, synchronization between data passing from dsp to app processor. I am hoping these changes can help to fix this issue. 
I am not able to reproduce it on my setup. Were you flying the drone when you see the issue? I need help to reproduce it if you are still seeing this issue.

wangxdflight, 2017-01-26:
Hi, Lorenz: is this issue still reproducible? I suspect the randomness was
due to lock/synchronization issue in shared memory, and it should have been
fixed with recent changes. Can you please confirm?
Thanks.

LorenzMeier, 2017-01-26:
@Stifael will be able to comment Monday

Stifael, 2017-01-26:
@wangxdflight:
I apologize for not returning to this issue earlier. Today I received a snapdragon which I havent touched yet. So tomorrow I will reflash that one a will comment the outputs here

Stifael, 2017-01-30:
@wangxdflight:
I am sorry for the delay. i tried to test it, but it did not work yet as expected. Did not see any spikes, but wasnt able to calibrate. I need to investigate farther.

wangxdflight, 2017-01-30:
I am seeing some issues which might be related updates in sensors.cpp,
voted_sensors_update.cpp. There was lots of changes introduced there.
Still debugging.

LorenzMeier, 2017-02-02:
@wangxdflight Any updates on this? I'd love to help if I can with general feedback.

wangxdflight, 2017-02-02:
Those sensor issues were root caused to be the timer inaccuracy. With the
update in driverframework, they are not observed anymore.
I haven't been able to reproduce the spiking issue.

LorenzMeier, 2017-02-06:
@Stifael Can you re-try latest PX4/master? I merged the changes from @wangxdflight.

Stifael, 2017-02-07:
@LorenzMeier @wangxdflight 
I just tested the latest master with one of the snappies. I tested calibration with qgc and I can confirm that this time no spikes were present and calibration proceeded as expected.

closed 2017-02-07 by Stifael