PX4/PX4-Autopilot #6370: Mocap Timeout Error

mohradza, 2017-01-17:
I am trying to use Vicon mocap data to supplement the LPE for use in position control. My setup is as follows: Pixhawk (FW version 1.5.1) connected via usb to a Tegra TK1 which runs the mavros script. Then I have a ground station computer running Ubuntu 16 through Parallels. The ground station runs the vicon_bridge script (which won't run on the TK1 because the processor is ARM based), and forwards the pose to /mavros/mocap/tf. The actual system is a flame wheel hex rotor.

Once I run the bridge script, the Pixhawk status led starts to breathe green. However, in QGroundControl, I am getting a mocap timeout message. I have checked that the messages are being sent fast enough, and I even tried throttling down the publish rate as suggested in this thread: https://github.com/PX4/Firmware/issues/5174. I am also getting hard sync error from Mavros. I can also confirm that the local_position/pose topic reflects the values I am seeing in the mocap/tf topic. However, when I try to do position hold it is definitely not working properly.

Any help is appreciated, thanks!

mhkabir, 2017-01-18:
If you're continually getting timesync errors and timeouts, you definitely have bandwidth issues. Practically every single one of the people in the thread you linked had hardware issues or link problems.
Check your connections. Try printing the deltas using printfs and see what sort of values you get.

mohradza, 2017-01-18:
I think the issue comes from not being able to run the vicon_bridge node on the same computer running the roscore / mavros, which in this case in the onboard computer. I have to run the vicon_bridge node from my laptop, which I suspect is introducing some lag when it has to communicate with the onboard computer over wifi. I will try to print out the deltas tomorrow to see what I get.

mhkabir, 2017-01-18:
Closing as not flightstack issue. Feel free to continue discussion on discuss.px4.io!

closed 2017-01-18 by mhkabir