PX4/PX4-Autopilot #6371: I would like an offboard mode kill switch

lovettchris, 2017-01-17:
I just realized the "offboard switch" is actually an "offboard mode on" switch, which is not what I was expecting, or needing.  What I really want for safe experimentation with offboard control is an "offboard kill" switch that stops offboard mode and moves the drone to "loiter" mode.  Interestingly I discovered that toggling my normal "flight mode" switch does not stop offboard mode either.  So either that is a bug, or I need an offboard kill switch.   So would it make sense for me to add this?  If so, should I add a new switch or should I take over the existing offboard switch code for this purpose ?  I personally don't understand how an "offboard on" switch is useful.  The software would have to be sending offboard commands before hand otherwise it will immediately time out and offboard will be cancelled, unless you disable that timeout, which doesn't seem safe either.  To me it makes more sense for the software to initiate offboard mode when it is ready, and for the switch to be a safety mechanism to quickly get the drone out of offboard mode.  This would then be sent back to the "offboard software" as a mode change which would signal it to stop sending offboard commands...

thoughts?

LorenzMeier, 2017-01-17:
In all use we've seen so far people were far more interested in enabling the off board mode from a switch than just canceling it. The only change you would need is to switch out of off board if any flight mode changes. That seems to make sense (and at that point you don't need an off board switch - you can switch to mission, engage off board and flick back to position control if things go wrong. Anybody else can then still use the dedicated switch.

I'm closing this as its not a real issue and I'm assuming you will send a PR which ends off board on mode change.

closed 2017-01-17 by LorenzMeier

lovettchris, 2017-01-18:
ok, will do.