PX4/PX4-Autopilot #6374: jMAVSim doesn't fly any more

lovettchris, 2017-01-18:
with latest code, using LPE (not EKF2), jMavSim won't get off the ground.  I debugged this down to line 549 of pwm_out_sim.cpp, I found that changing this:

 			_lockdown = aa.lockdown || aa.manual_lockdown;

to this:

 			_lockdown = aa.manual_lockdown;

solves the problem.  The reason this is needed is because state_machine_helper.cpp line 196 when it is asked to enter HIL_STATE_ON sets lockdown to true.

		if (status->hil_state == vehicle_status_s::HIL_STATE_ON) {
			armed->lockdown = true;

Therefore no PWM output was being generated as a result...  should I submit this fix as a pull request?

LorenzMeier, 2017-02-16:
Yep that is indeed a bug.

LorenzMeier, 2017-02-16:
Fixed in 7bd851469796f16d51938c318f2b45dd0e2201e0.

closed 2017-02-16 by LorenzMeier