PX4/PX4-Autopilot #6383: Daily test flight log analysis

LorenzMeier, 2017-01-18:
http://review.px4.io/plot_app?log=90a3f19b-c8c2-44db-887f-0e6f5488aa84

@priseborough We should analyse this and determine root cause.

LorenzMeier, 2017-01-18:
@MaEtUgR @Stifael Please review first thing in the morning. This could be a throttle feedforward fail.

priseborough, 2017-01-19:
I started with looking at the estimator outputs and have confirmed that innovations are within limits up until ground impact at 354.35 sec.

I see the following series of events:

1) At 348.4 sec there is a switch to attitude control
2) From 350.9 to 351.5 seconds the throttle stick is increased from around 50% to 100% and the rate controller thrust setpoint follows it, achieving about 75% throttle.
3) From 351.7 to 352.0 seconds, the throttle stick reduces from 100% to 25% and the rate controller thrust setpoint also reduces from 75% to 13%. Simultaneous with this reduction in pilot demanded thrust, the pilot demands a roll input of 25%.
4) At 352.0 seconds, due to the inadequate thrust, the copter loses control of its roll rate and flips. When this happens the thrust setpoint from the attitude controller goes negative.
5) From 352.2 to 352.9 seconds the pilot reacts to the loss of attitude control and throttles up, but the attitude controller continues to provide a negative thrust setpoint, despite the pilot requesting maximum thrust.

This highlights some fundamental deficiencies in the attitude control loops because:

a) The pilot should always have control over thrust in attitude mode and,
b) If the quad is inverted in a non pilot throttle mode, eg altitude, position, etc, then the throttle should be set to 50% to maximise control moments.

Stifael, 2017-01-19:
I just had a look, and actually I am struggling to see to what mode the copter switches. The description of the log says that it jumps from non-gps to gps. this is at the beginning of the flight when switching to mission. the undesired delay happens during mission. and the description says that the crash occurs during altitude mode, but I am not sure about that since the color of the flight review plot are too similar. can we see the modes from the Flightplot app somehow?

Stifael, 2017-01-19:
@priseborough thank you for the good analysis.
I tried to take pictures from what paul said to better see whats happening:
1. switch to altitude control
2. ramp up throttle (channel 2)-> increase thrust
[image]
[image]
3. ramp down throttle and ask for roll -> thrust decreases
[image]
[image]
4. flip
5. throttle up, due to flip thrust becomes more negative
[image]
[image]

I think point 4. and 5. is not surprising, since If I am correctly remember then the attitude controller was not even designed for inverted orientation (correct me if I am wrong). About the situation where the thrust is low and the user has not full control over attitude (due to the low thrust) is probably the first thing that needs to be changed.

LorenzMeier, 2017-01-19:
The attitude controller is full 3D and does handle inversion correctly (at least it did before the last set of changes). 

Check out the beginning of this video:
https://vimeo.com/85110759

LorenzMeier, 2017-01-19:
@Stifael The low throttle setting is fine - if its not working then its due to incorrect ESC calibration / PWM MIN settings for this particular setup.

Stifael, 2017-01-19:
yep, I mean if the tilt attitude is set to conservative values, then the low throttle setting should work, but that does not mean that at some point we should change that such that you still can control attitude when low thrust is required. IMHO, that would be an improvement.
About 3D: I know the position controller is not 3D (there are parts in the code that do not work if flipped), and since it was in altitude control, the flip might be caused by the z pos control.

priseborough, 2017-01-19:
So how are users to know how much min PWM is required to right a copter? If I was inverted in manual mode, I wouldn't want the controller reducing my throttle to minimum and taking away control authority, particularly given that when falling inverted with non-braking esc's the inflow can make small  PWM changes around idle ineffective due to the free-wheel back emf.

LorenzMeier, 2017-01-19:
@priseborough It would automatically give a minimum amount of throttle and boost it when hitting mixer limits. The trick is to make sure that what we think is 30% is in fact 30% of throttle and not 20%.

priseborough, 2017-01-19:
@LorenzMeier In this instance all esc demands are pegged against the minimum value, despite the pilot throttle demand being at maximum, so regardless of esc calibration the controller was failing to generate esc demands with a righting moment.

LorenzMeier, 2017-01-19:
Yep. Regression from a recent improvement - plain sign error.

closed 2017-01-19 by LorenzMeier