PX4/PX4-Autopilot #6388: EKF2 performance poor in gazebo SITL with GPS

jgoppert, 2017-01-19:
Here is a 3-way comparison of EKF2, IEKF, and LPE for a SITL box mission with varying altitude. EKF2 fails on the first turn even before the climb. I think we might want to add an EKF2 multirotor test mission. I know the original thinking was that it was some server cpu limitation, but I can't get it to work locally either. Is anyone testing EKF2 in the wild with mulitrotors and can they confirm this is just on the simulation side?

EKF2: http://logs.px4.io/plot_app?log=bd702929-6e9c-452f-b2b3-398e53c9cfb3

[image]

IEKF: http://logs.px4.io/plot_app?log=6c99fd56-0863-4e24-bf72-6b67adb27392

[image]

LPE: http://logs.px4.io/plot_app?log=3fbb051c-aeb8-44af-bff3-459c4e771fb1

[image]

tumbili, 2017-01-19:
@jgoppert What kind of gps delay are you using in your tests and do we know the delay that is added in gazebo? I remember that I once opened an issue because gazebo did not delay the gps signal.

jgoppert, 2017-01-19:
Yes, that is true. Currently sitl gps is undelayed and has no noise. We should look into fixing that.  Shouldn't be too hard.

jgoppert, 2017-01-19:
I add GPS delay and noise to sim here: https://github.com/PX4/sitl_gazebo/pull/82

jgoppert, 2017-05-01:
Resolved.

closed 2017-05-01 by jgoppert