PX4/PX4-Autopilot #6398: NuttX config maintenance

lucasdemarchi, 2017-01-20:
I asked this on gitter some time ago with not responses and today we were bit by it with the aerofc board.
See https://github.com/PX4/Firmware/pull/6397 

How are the nuttx configuration files for the different boards maintained?

I thought the `reconfigure_nuttx_*` target was supposed to do that, but it doesn't work. Removing the call to "make oldconfig" works, but it produces a configuration so different that I wonder if this is really the way you are doing it.

And talking about different configuration, if I do this it would contain the fix from the PR above, but just because that's the default value for that entry.

Trying for another board like px4fmu-v2 didn't help: lots of difference from what's in tree.

lucasdemarchi, 2017-01-20:
@davids5 @LorenzMeier see above

davids5, 2017-01-20:
See https://github.com/PX4/Firmware/pull/6397#issuecomment-273943550 for directions.

closed 2017-01-20 by davids5