PX4/PX4-Autopilot #6399: Integration test on master fails

LorenzMeier, 2017-01-20:
@bkueng One of your recent ulog changes seemed to have broken the CI test in some way.

```
[ROSTEST]-----------------------------------------------------------------------

[px4.rosunit-mission_test_new_1/test_mission][passed]
[px4.rosunit-mission_test_new_2/test_mission][ERROR]----------------------------
right keys must be sorted
 File "/usr/lib/python2.7/unittest/case.py", line 331, in run
   testMethod()
 File "/job/catkin/src/px4/integrationtests/python_src/px4_it/mavros/mission_test.py", line 372, in test_mission
   data = px4tools.ulog.read_ulog(last_log).concat(dt=0.1)
 File "/usr/local/lib/python2.7/dist-packages/px4tools/ulog.py", line 356, in concat
   m = pd.merge_asof(m, df, 'timestamp')
 File "/usr/local/lib/python2.7/dist-packages/pandas/tools/merge.py", line 471, in merge_asof
   return op.get_result()
 File "/usr/local/lib/python2.7/dist-packages/pandas/tools/merge.py", line 968, in get_result
   join_index, left_indexer, right_indexer = self._get_join_info()
 File "/usr/local/lib/python2.7/dist-packages/pandas/tools/merge.py", line 714, in _get_join_info
   right_indexer) = self._get_join_indexers()
 File "/usr/local/lib/python2.7/dist-packages/pandas/tools/merge.py", line 1180, in _get_join_indexers
   raise ValueError('right' + msg)
--------------------------------------------------------------------------------
```

jgoppert, 2017-01-20:
pandas is using merge_asof, it requires timestamp to be monotonic. The work around is to use this: http://pandas.pydata.org/pandas-docs/stable/generated/pandas.DataFrame.sort_values.html

I can force px4tools to sort on timestamp if you keep having an issue.

LorenzMeier, 2017-01-20:
It has been flaky for some time so it would be fantastic if you could do the change.

jgoppert, 2017-01-20:
Here is the patch: https://github.com/dronecrew/px4tools/commit/be7fac5cf49e046d3dca61cb9e82a6e4c2928657

I pushed it to pip and released it with tag v0.7.8

bkueng, 2017-01-20:
@jgoppert It should indeed be monotonic. Which topic is affected?

jgoppert, 2017-01-20:
@bkueng If you can find a log you know it broke on I can tell you by catching the exception and printing the topic name.

jgoppert, 2017-01-20:
I pushed a fix for this: http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL%20PRs/3991/console it should be live on pip soon. https://github.com/dronecrew/px4tools/commit/a4d644bb2c042da0c20d53040e817963e82d74d9

dagar, 2017-04-11:
Fixed.

closed 2017-04-11 by dagar