PX4/PX4-Autopilot #6402: Cannot compile sourcecode with mavlink v1 headers

aysh-shariff, 2017-01-20:
I am trying to build the sourcecode with mavlink v1. I modified the mavlink_bridge_header.h  to include
#include <v1.0/mavlink_types.h
#include <v1.0/common/mavlink.h

But I get redefinition errors. How do I fix them ?

Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_mission.cpp.obj
[ 76%] In file included from /home/malte/px4_current/src/Firmware/src/modules/mavlink/mavlink_log_handler.h:44:0,
                 from /home/malte/px4_current/src/Firmware/src/modules/mavlink/mavlink_main.h:71,
                 from /home/malte/px4_current/src/Firmware/src/modules/mavlink/mavlink_main.cpp:86:
/home/malte/px4_current/src/Firmware/mavlink/include/mavlink/v2.0/mavlink_types.h:23:0: error: "MAVLINK_CORE_HEADER_LEN" redefined [-Werror]
 #define MAVLINK_CORE_HEADER_LEN 9 ///< Length of core header (of the comm. layer)
 ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
Built target lib__external_lgpl
make[3]: *** [src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_main.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
[ 77%] Built target modules__mc_pos_control
[ 77%] Built target modules__mc_att_control
[ 78%] Built target modules__sdlog2
In file included from /home/malte/px4_current/src/Firmware/src/modules/mavlink/mavlink_log_handler.h:44:0,
                 from /home/malte/px4_current/src/Firmware/src/modules/mavlink/mavlink_main.h:71,
                 from /home/malte/px4_current/src/Firmware/src/modules/mavlink/mavlink_mission.cpp:44:
/home/malte/px4_current/src/Firmware/mavlink/include/mavlink/v2.0/mavlink_types.h:23:0: error: "MAVLINK_CORE_HEADER_LEN" redefined [-Werror]
 #define MAVLINK_CORE_HEADER_LEN 9 ///< Length of core header (of the comm. layer)
 ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[ 79%] Built target modules__vtol_att_control
make[3]: *** [src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_mission.cpp.obj] Error 1
make[2]: *** [src/modules/mavlink/CMakeFiles/modules__mavlink.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 79%] Built target modules__param
[ 82%] [ 82%] Built target modules__systemlib__mixer
Built target modules__uORB
make[1]: *** [all] Error 2
make: *** [px4fmu-v2_default] Error 2

dagar, 2017-01-20:
Hi @aysh-shariff,

It's not immediately clear to me what's specifically wrong, or how you have this organized. Did you generate both mavlink v1 and v2 and place the output in the same location that your firmware source uses?

If you go into Firmware/mavlink/include/mavlink/v1.0 (or v2.0) and run `git status` you should see changes that correspond to your message addition.

aysh-shariff, 2017-01-20:
@dagar git status does show header files pertaining to my msgs in the untracked files.

dagar, 2017-01-23:
Try generating mavlink/v1.0 *and* mavlink/v2.0 and restore mavlink_bridge_header.h to its original form. The mavlink module supports mavlink versions 1 and 2. See param MAV_PROTO_VER.

dagar, 2017-01-25:
@aysh-shariff is this resolved?

mhkabir, 2017-01-25:
Closing as resolved. There's no need to edit anything in the source. Just the headers need to be generated and placed in the directories.

closed 2017-01-25 by mhkabir

aysh-shariff, 2017-01-25:
@dagar I generated both the headers and placed in the respective folders. 
However, using 
#include <v1.0/mavlink_types.h
#include <v1.0/common/mavlink.h 

still gives compilation errors. But I am using v2 headers in mavlink_bridge_header.h since both the headers are compatible. That solves my issue. So I'm not concerned about the compilation errors anymore.