PX4/PX4-Autopilot #6403: Getting run_tests.bash working with rotors_simulator (gbmhunter/rotors_integration).

gbmhunter, 2017-01-20:
@jgoppert I see you've contributed a fair amount to `run_tests.bash`, I was wondering if you could help me with the following error when running `bash integrationtests/run_tests.bash`:

````
<build messages here ...>
Traceback (most recent call last):
  File "/opt/ros/indigo/bin/rostest", line 36, in <module>
    rostestmain()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/__init__.py", line 273, in rostestmain
    _main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/rostest_main.py", line 159, in rostestmain
    testCase = rostest.runner.createUnitTest(pkg, test_file, options.reuse_master, options.clear, options.results_base_dir)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/runner.py", line 229, in createUnitTest
    config = roslaunch.parent.load_config_default([test_file], None)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/config.py", line 453, in load_config_default
    raise RLException(e)
roslaunch.core.RLException: while processing /home/gbmhunter/src/catkin/src/px4/launch/mavros_posix_sitl.launch:
Invalid <arg> tag: rotors_gazebo_plugins
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/gbmhunter/src/catkin/src
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks. 

Arg xml is <arg default="$(find rotors_gazebo_plugins)/worlds/empty.world" name="world"/>
````

This occurs after switching the `sitl_gazebo` submodule out for the new and improved `rotors_simulator`. I am currently working on this at https://github.com/gbmhunter/Firmware/tree/rotors_integration

jgoppert, 2017-01-20:
This is ROS saying that it cannot find the rotors_gazebo_plugins package in the workspace. It creates a catkin workspace in the directory here: home/gbmhunter/src/catkin/src, so take a look. Try to source the ./devel/setup.bash. Then see if ros can run anything from the package  with roslaunch or rosrun.

gbmhunter, 2017-01-20:
Ah thanks for the tip r.e. the catkin workspace.
`roslaunch rotors_gazebo ..`. seems to be valid syntax, but not
`roslaunch rotors_gazebo_plugins ...`
(both are packages within the rotors_simulator repo)

This may make sense as the `rotors_gazebo` package has the launch files. Still investigating...

jgoppert, 2017-01-20:
try running catkin build and then rerun ./devel/setup.bash

It might just be running catkin build px4, in which case you need to add all dependencies to Firmware/package.xml

gbmhunter, 2017-01-20:
If I try and run `catkin make` I get this error:
````
The build space at '/home/gbmhunter/src/catkin/build' was previously built by 'catkin_make'. Please remove the build space or pick a different build space.
````

`catkin clean` reports that it can't execute because it is not a valid workspace.

jgoppert, 2017-01-20:
I use catkin tools. Not catkin make. See integration script for how to use it. Catkin make is not compatible.

gbmhunter, 2017-01-20:
Ah cool, `catkin_make` works fine.

gbmhunter, 2017-01-20:
@jgoppert I get past that error if I create symlinks in `run_tests.bash` to the rotors_gazebo and mav_msgs packages (both contained in sub-folders of different submodules within PX4):
````
ln -s $ORIG_SRC/Tools/mav_comm/mav_msgs ${CATKIN_DIR}/src/mav_msgs
ln -s $ORIG_SRC/Tools/rotors_simulator/rotors_gazebo ${CATKIN_DIR}/src/rotors_gazebo
````

Now I get a similar error, but this time for mavros. This has got me a little more stuck as this is a completely separate repo (https://github.com/mavlink/mavros) from `sitl_gazebo`, and I don't I believe I have touched any code involving it, so I'm trying to work out why it would be missing...

````
roslaunch.core.RLException: while processing /home/gbmhunter/src/catkin/src/px4/launch/mavros_posix_sitl.launch:
Invalid <arg> tag: mavros
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/gbmhunter/src/catkin/src
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks. 

Arg xml is <arg default="$(find mavros)/launch/px4_pluginlists.yaml" name="pluginlists_yaml"/>
````

I don't feel like going and adding this as a submodule to PX4 and making symlinks is the right thing todo here. I notice many of the launch files already depend on mavros.

gbmhunter, 2017-01-23:
O.K., most of these problems have been resolved by running the commands inside the `px4-dev-ros:v1.0` docker image and doing a few changes to the load script.

I am now able to run these tests on my local system. Now it seems like all the tests are just timing out (there is 3 of them?). What exactly is needed for the test to pass/complete at this point?

gbmhunter, 2017-01-23:
@jgoppert @LorenzMeier The `mission_test.py` test seems to be getting stuck waiting for a global position, as shown by:

[image]

gbmhunter, 2017-01-30:
Issues here have been resolved.

closed 2017-01-30 by gbmhunter