PX4/PX4-Autopilot #6404: Mavlink messages do not get relayed to GCS

sanderux, 2017-01-20:
It seems when sending a mavlink statustext message from a board computer PX4 does not relay the message to the ground station. The question is if this should be implemented and how.

Considerations would be the use of a target_system and default behavior when such an ID is omitted.

In dronekit i tried:
```
msg = vehicle.message_factory.statustext_encode(6,"Test message")
vehicle.send_mavlink(msg)
```
I also tried
```
vehicle._master.mav.statustext_send(6,"Test message")
```
Neither works

AndreasAntener, 2017-01-21:
Did you try setting the forwarding flag `-f` on the mavlink instance that has telemetry attached?

sanderux, 2017-01-21:
The is no active management on that side, it's just relaying serial to UDP

AndreasAntener, 2017-01-22:
I'm not sure what you are saying. How does the setup look?

sanderux, 2017-01-22:
I'll pm you

mhkabir, 2017-04-03:
Closing as devcall discussion indicated that this is not an issue.

closed 2017-04-03 by mhkabir