PX4/PX4-Autopilot #6408: Change PX4 SITL MAVLink interface port number

anuppari, 2017-01-20:
I'm trying to run a simulation with multiple quadrotors in Gazebo with ROS. I think I can just spawn multiple gazebo models and corresponding px4 nodes in separate namespaces, however they will all publish MAVLink messages to the same port number, and each instance of mavros won't be able to distinguish messages from the individual px4 nodes.

Does the px4 node (i.e., the one launched [here](https://github.com/PX4/Firmware/blob/7aea2ca0308295b2e2dca30f758a17b6ef34e759/launch/posix_sitl.launch#L24-L26)) take a port number parameter, or is there anywhere else I can specify port numbers?

Thanks!

anuppari, 2017-01-20:
Also, if this feature is not currently implemented, I can probably add it myself if you could point me to the correct source file for that node. The directory structure is different than the typical ROS packages that I am used to and I'm having a hard time trying to figure out which source files generate the ROS nodes.

Thanks

anuppari, 2017-01-24:
I'll share my progress in case anyone is trying to do something similar.

The files in [this folder](https://github.com/PX4/Firmware/tree/master/posix-configs/SITL/init) are the startup scripts for the simulated px4 (e.g., [this file](https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/lpe/iris) is the default script), in which the port parameters for mavlink are set.

Starting up multiple instances of the px4 in there own ROS namespaces works fine as far as ROS and Gazebo are concerned, however the quads in the simulation fly erratically. I think this is because the uORB topics are not grouped into their own namespace, and so the two px4's are interfering with each other. If only 1 px4 is launched (even within a namespace), everything works perfectly. Looking through the uORB code, there doesn't seem to be an easy way to push all the topic names down into a namespace like can be done in ROS launch files.

If anyone more familiar with uORB has any suggestions, it would be greatly appreciated.

bkueng, 2017-01-24:
Hi @anuppari

uORB is used for intra-process communication on Linux. What you need is to start multiple `px4` processes and make sure they use different ports for the mavlink communication with the simulator.
I haven't used it, but there is a script for this here: https://github.com/PX4/Firmware/blob/master/Tools/sitl_multiple_run.sh. Not sure if that still works though.

anuppari, 2017-01-24:
I think I'm doing something equivalent to the script, but through roslaunch.

Here is the main launch file I call (`asap_gazebo_sim` is the name of my package):
```xml
<launch>
    <!-- Start Gazebo world -->
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="headless" value="false"/>
        <arg name="gui" value="true"/>
        <arg name="world_name" value="$(find mavlink_sitl_gazebo)/worlds/empty.world" />
        <arg name="debug" value="false" />
        <arg name="verbose" value="false" />
        <arg name="paused" value="false" />
    </include>

    <!-- Start first instance of px4 and gazebo model-->
    <group ns="uas1">
        <include file="$(find asap_gazebo_sim)/launch/quad.launch">
            <arg name="name" value="uas1" />
            <arg name="rcS" value="$(find asap_gazebo_sim)/px4-configs/lpe/iris1"/>
            <arg name="fcu_url" value="udp://:14540@localhost:14557"/>

            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
        </include>
    </group>

    <!-- Start second instance of px4 and gazebo model-->
    <group ns="uas2">
        <include file="$(find asap_gazebo_sim)/launch/quad.launch">
            <arg name="name" value="uas2" />
            <arg name="rcS" value="$(find asap_gazebo_sim)/px4-configs/lpe/iris2"/>
            <arg name="fcu_url" value="udp://:14541@localhost:14559"/>

            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
        </include>
    </group>
</launch>
```

The `quad.launch` file is:

```xml
<launch>
    <!-- Parameters -->
    <arg name="name" default="uas" />
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>

    <arg name="est" default="lpe"/>
    <arg name="vehicle" default="iris"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>

    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)"/>
    <arg name="fcu_url" default="udp://:14540@localhost:14557"/>

    <!-- px4 instance -->
    <node name="sitl" pkg="px4" type="px4" output="screen" args="$(find px4) $(arg rcS)" />

    <!-- load model in gazebo -->
    <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-sdf -file $(arg sdf) -model $(arg name) -x $(arg x) -y $(arg y) -z $(arg z) -R 0 -P 0 -Y 0"/>

    <!-- mavros instance for ROS communication -->
    <include file="$(find mavros)/launch/px4.launch">
        <arg name="fcu_url" value="$(arg fcu_url)" />
    </include>
</launch>
```

The `iris{n}` files called in the main launch file are the same as the [file here](https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/lpe/iris) except with the mavros ports changed:
iris1:
```
...
mavlink start -u 14556 -r 4000000
mavlink start -u 14557 -r 4000000 -m onboard -o 14540
mavlink stream -r 50 -s POSITION_TARGET_LOCAL_NED -u 14556
mavlink stream -r 50 -s LOCAL_POSITION_NED -u 14556
mavlink stream -r 50 -s GLOBAL_POSITION_INT -u 14556
...
```
iris2:
```
...
mavlink start -u 14558 -r 4000000
mavlink start -u 14559 -r 4000000 -m onboard -o 14541
mavlink stream -r 50 -s POSITION_TARGET_LOCAL_NED -u 14558
mavlink stream -r 50 -s LOCAL_POSITION_NED -u 14558
...
```

So from my understanding, I am starting multiple instances of px4, each with their own mavlink ports. Though I did notice that when I run `rosnode list` I don't see any px4 nodes, even when only launching one instance.

I also noticed that when launching 2 instances, I get a `WARN  [simulator] bind failed`, which is not present when I launch either of the individual instance (i.e., only launch uas1 or uas2). In this case, I can run `commander takeoff` and both instances attempt to takeoff, though one inevitably flips over and crashes, and the other flies around erratically. Also in this case, only one of the instances of mavros actually publishes data. Both instances are on completely different ports, so I don't know why binding would fail or why only one mavros works.

anuppari, 2017-01-24:
The erratic behavior leads me to believe there is some cross-talk somewhere. The ROS topics are all isolated, so I think it might be in the uORB topics. [This tutorial](https://dev.px4.io/tutorial-hello-sky.html#step-5-subscribing-sensor-data) seems to suggest that there is only one global topic for each msg type, unless you include modifications presented [here](https://dev.px4.io/advanced-uorb.html#adding-a-new-topic). However, [most of the msg definitions](https://github.com/PX4/Firmware/tree/master/msg/) do not include a `# TOPICS` line, so the multiple instances of px4 might be communicating through the same channels.

From my understanding, uORB uses a file abstraction (e.g., [this page](https://pixhawk.org/zh/dev/shared_object_communication) explains that the subscriber handles are file descriptors), so is it possible that the multiple instances of uORB are reading/writing to the same "files"?

nicolaerosia, 2017-01-24:
Gazebo plugin is communicating with PX4 `simulator` module on a default port 14560.
Relevant links:
https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris#L47
https://github.com/PX4/Firmware/blob/master/src/modules/simulator/simulator.cpp#L164

anuppari, 2017-01-24:
Thanks @nicolaerosia, that fixed the `WARN [simulator] bind failed` issue, but the quads still fly erratically. Any clarification on whether uORB cross-talk is a potential cause?

nicolaerosia, 2017-01-24:
You're welcome!
I would say no since I believe the IPC is done in px4 private memory.
You need to make sure you bind the Gazebo instance of the quad with the corresponding PX4 simulator.

anuppari, 2017-01-24:
Thanks, that fixed it.

For anyone else interested, the additional changes I made were adding the port argument (i.e., `-u [port #]` to the px4 startup script (e.g., [this line](https://github.com/PX4/Firmware/blob/523688e43c4563ef92b329e05d21a2df6a3b0c47/posix-configs/SITL/init/ekf2/iris#L47) @nicolaerosia linked above) as well as change the port number for the corresponding gazebo plugin (e.g., the `mavlink_interface` plugin in the sdf file that is generated in [this folder](https://github.com/PX4/sitl_gazebo/tree/3512ec89592feb6b14c791a09cc3476146f0de24/models/iris). Here's the corresponding line for the [solo model](https://github.com/PX4/sitl_gazebo/blob/3512ec89592feb6b14c791a09cc3476146f0de24/models/solo/solo.sdf#L430)).

By the way, the base xacro file used to generate the iris sdf has a small typo. The arguments defined on [L7-L8](https://github.com/PX4/sitl_gazebo/blob/3512ec89592feb6b14c791a09cc3476146f0de24/models/rotors_description/urdf/iris_base.xacro#L7-L8) are not used in the macro on [L35-L36](https://github.com/PX4/sitl_gazebo/blob/3512ec89592feb6b14c791a09cc3476146f0de24/models/rotors_description/urdf/iris_base.xacro#L35-L36).

LorenzMeier, 2017-01-24:
Would you mind sending a pull request?

anuppari, 2017-01-24:
No problem, see https://github.com/PX4/sitl_gazebo/pull/83

Doolly, 2017-01-26:
what port number do I have to use in 
<arg name="fcu_url" value="udp://:14560@localhost:14559"/>
and 
mavlink start -u 14559 -r 4000000 -m onboard -o 14560 ?
I get 'unsupported FCU' message or 'address already in use' messages when I change port numbers in variable ways

dennisss, 2017-01-26:
Should anyone find it useful, I've written a gazebo plugin (https://github.com/dennisss/tansa/blob/master/src/gazebo/swarm_plugin.cpp) for hot-loading models into an open gazebo world at arbitrary positions and starting the SITL instances. The instance starting is based on the previously mentioned sitl_multiple_run.sh modified to work with the latest master (https://github.com/dennisss/tansa/blob/master/scripts/start_sim.sh).

anuppari, 2017-01-26:
@Doolly, as [mentioned here](https://github.com/PX4/Firmware/issues/6408#issuecomment-274886350), port 14560 is already used for communications from gazebo to the simulator/px4, so you need to use different ports for communication between px4 and ROS. Your two lines are correct in the sense that it should be:

arg name="fcu_url" value="udp://:{port1}@localhost:{port2}"
and
mavlink start -u {port2} -r 4000000 -m onboard -o {port1}

anuppari, 2017-01-26:
Can anyone explain why there are two mavlink streams, i.e., in iris there is:
```
mavlink start -u 14556 -r 4000000
mavlink start -u 14557 -r 4000000 -m onboard -o 14540
```

What is the first stream on port 14556 for? Currently each instance requires keeping track of, and editing files, with 4 different ports, I'm hoping to streamline this a bit if possible.

weiweikong, 2017-01-27:
@anuppari There is one port for connection between Gazebo and PX4 module, and the other is for connection between PX4 module with ROS.

nicolaerosia, 2017-01-27:
@anuppari IMO The first port is for Ground Control Station communication, while the second one, "onboard" is for the companion computer, which in your case is ROS
See here:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1810
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1986

lenny521, 2017-04-24:
Hello, have your problem be solved ？ i would like to consult you how to run multiple UAV in gazebo, look forward to your reply, best wishes!

paan2097, 2017-04-28:
Gazebo defaults to mavlink udp port 14560. https://github.com/PX4/sitl_gazebo/blob/a8f57664c7bf6f4d8ab7a0c01d65181455151ca0/include/gazebo_mavlink_interface.h#L55

PX4 starts streaming mavlink here (This was also declared in solo.sdf files): https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/plane#L72

paan2097, 2017-04-28:
@dennisss Can you give instructions on how to run after running sitl_multiple_run.sh please?

AliSiddiqui4, 2017-05-07:
@anuppari @nicolaerosia  any idea how can I do the same for the erle copter [tutorial](http://docs.erlerobotics.com/simulation/vehicles/erle_copter/tutorial_1). I have managed to spawn multiple erle copters in Gazebo but as there is only one MAVlink that links to them so they fly away erratically. Please help, I am new to this so a step by step solution would be appreciated.

dennisss, 2017-05-07:
If anyone wants a turn key solution to this, I implemented a solution to this in my research platform here (https://github.com/dennisss/tansa/tree/master/doc/examples/gazebo_spawn), given a list of positions, an rcS file, and a single sdf file, it will programmatically hot load models into the world. It takes about 10 lines of code to use. This has substantially sped up our development speed.

This works by having a gazebo plugin (https://github.com/dennisss/tansa/blob/master/src/gazebo/swarm_plugin.cpp) which modifies the position, port assignments etc. of a single sdf in order to adapt it to many unique instances.

It is accompanied by a helper class on the client side (https://github.com/dennisss/tansa/blob/master/src/gazebo/gazebo.cpp#L141) which sends protobuf messages to the plugin to request new drones to be spawned / respawned.

PX4 instances are created using a templated rcS file like https://github.com/dennisss/tansa/blob/master/config/models/x340/rcS and a modified version of stitl_multiple.run.sh here https://github.com/dennisss/tansa/blob/master/scripts/start_many_instances.sh . 

Feel free to adapt to your particular projects.

AliSiddiqui4, 2017-05-08:
@dennisss I tried your solution, following the steps I tried to make build. First I got the erro to update CMake , after resolving that I got error to update BOOST and after removing all my packages I update boost to 1.55 I am still getting this error: 

```
make[3]: Entering directory `/home/muhammad/simulation/tansa/build_socketio_cpp'
make[3]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libboost_random.so', needed by `libsioclient.so.1.6.0'.  Stop.
make[3]: Leaving directory `/home/muhammad/simulation/tansa/build_socketio_cpp'
make[2]: *** [CMakeFiles/sioclient.dir/all] Error 2
make[2]: Leaving directory `/home/muhammad/simulation/tansa/build_socketio_cpp'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/muhammad/simulation/tansa/build_socketio_cpp'
make: *** [build_socketio_cpp] Error 2

```

dennisss, 2017-05-08:
@AliSiddiqui4 assuming you are on Ubuntu, you can try doing `sudo apt-get install libboost-random-dev` or with `libboost-random1.55-dev`. If that doesn't work, let me know which version of the OS you are on.

AliSiddiqui4, 2017-05-08:
@dennisss ok that helped but I am getting another error now: 

`mkdir -p build
rm -f config/gazebo/models/x340/x340.sdf
cd build; cmake ..; make
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   system
--   filesystem
CMake Error at cmake/FindEigen3.cmake:39 (file):
  file failed to open for reading (No such file or directory):

    /usr/include/eigen3/Eigen/src/Core/util/Macros.h
Call Stack (most recent call first):
  cmake/FindEigen3.cmake:65 (_eigen3_check_version)
  CMakeLists.txt:10 (find_package)

CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "FindCGAL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "CGAL", but
  CMake did not find one.

  Could not find a package configuration file provided by "CGAL" with any of
  the following names:

    CGALConfig.cmake
    cgal-config.cmake

  Add the installation prefix of "CGAL" to CMAKE_PREFIX_PATH or set
  "CGAL_DIR" to a directory containing one of the above files.  If "CGAL"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/muhammad/simulation/tansa/CMakeFiles/CMakeOutput.log".
make[1]: Entering directory `/home/muhammad/simulation/tansa/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory `/home/muhammad/simulation/tansa/build'
make: *** [build] Error 2
`

dennisss, 2017-05-08:
@AliSiddiqui4  please see https://github.com/dennisss/tansa/blob/master/config/docker/dev/Dockerfile for a full list of all the commands we use to install our system (Ignoring the RUN at the start of each line). There's a lot more in there than you probably need. I think you just need to do `sudo apt-get install libeigen3-dev libcgal-dev`

AliSiddiqui4, 2017-05-08:
@dennisss I am using ubuntu 14.04 maybe that's the problem. I can upgrade it but can you please elaborate on what this Swarm solution is? Can I easily simulate Swarm of Quads in Gazebo using it and write code for pattern formation and obstacle avoidance for it?

dennisss, 2017-05-08:
@AliSiddiqui4 I don't see why it wouldn't work on 14.04, if you use the same list of dependencies I don't think that much has changed between that and 16.04.

The spawning of models in gazebo is fairly easy with that example code. We have our own not yet well documented drivers for formation flight, although just about any other tool would work once they are spawned into gazebo. Similarly to sitl_multiple_run.sh in the PX4/Firmware repo, it will start every firmware instance for a unique mavlink instance sending packets to udp port 14550 + i*10, so just about any tool such as mavros or whatever that supports mavlink can be configured to listen on that port can used.

paan2097, 2017-05-08:
@dennisss Could you advise on how to add support for planes?

dennisss, 2017-05-08:
@paan2097 the only thing vehicle type specific is the rcS and sdf files. In my code I specify them in main.cpp

rcS file:
Pick an example such as one of these (https://github.com/PX4/Firmware/tree/master/posix-configs/SITL/init/ekf2) and template it so that it has the _MAINPORT_ syntax like (https://github.com/dennisss/tansa/blob/master/config/models/x340/rcS)

sdf file: these are a little bit more complicated, but sitl_gazebo has many examples. see the Tools/sitl_gazebo/models/plane/plane.sdf file with respect to the Firmware directory for one example.

If there is one that works well for you in a single instance for you, then these should be easy to find, although I haven't set up the tooling for finding them automatically yet.

paan2097, 2017-05-08:
@dennisss Thanks! Adding the sdf and rcS-like plane files in the same locations along with altering the mavlink linkes in rcS and adding two mavlink_udp_port likes form the sdf got it to work. Thanks!

AliSiddiqui4, 2017-05-09:
@dennisss I have built everything but stll unable to run the simulation: 
```
siddiqui@siddiqui-HP-G62-Notebook-PC:~/simulation/tansa$ make build
siddiqui@siddiqui-HP-G62-Notebook-PC:~/simulation/tansa$ make build_firmware
siddiqui@siddiqui-HP-G62-Notebook-PC:~/simulation/tansa$ npm install
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.1: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN extract-text-webpack-plugin@1.0.1 requires a peer of webpack@^1.9.11 but none was installed.
npm WARN material-ui@0.15.0 requires a peer of react-tap-event-plugin@^1.0.0 but none was installed.
npm WARN react-addons-transition-group@15.5.2 requires a peer of react@^15.4.2 but none was installed.
npm WARN webpack-dev-server@1.14.1 requires a peer of webpack@>=1.3.0 <3 but none was installed.
siddiqui@siddiqui-HP-G62-Notebook-PC:~/simulation/tansa$ make sim
make: *** No rule to make target 'sim'.  Stop.

```

dennisss, 2017-05-09:
@AliSiddiqui4 not sure why the make target isn't being recognized. You can try bypassing it and just running `./scripts/start_gazebo.sh` to start the simulator

mago3421, 2017-05-09:
@dennisss Thanks for providing this script. I have successfully instanced two planes. I'm not trying to connect to each plane to control it with Dronekit scripts. However, I do not know what the IP and port numbers would be for the connection strings to each plane. I've tried local loopback and my localhost addresses with both 14550 and 14560 but I get a timeout on dronekit. I also ran sockstat on each of the processes. Only the 14550 process occupies a UDP port according to sockstat. How can I find the connection IP/port addresses so that I may connect and control the planes via Dronekit?

dennisss, 2017-05-09:
@mago3421 the ports are configured for an instance right here (https://github.com/dennisss/tansa/blob/master/scripts/start_instance.sh#L25) to be 14550 and 14560, and that's what has been working for me on localhost. Each instance spits out error logs into `tmp/vehicle_X` directory, so please confirm that there are no errors being reported there. I sometimes get the issue that the firmware is not compiled and a run of `make build_firmware` is sometimes required

paan2097, 2017-05-09:
@dennisss So I copied over the plane sdf and rcS file and transfered the MAINPORT lines. but the two instances stop on Please start the flight simulator to proceed, even though I'm running make runSim and ./spawner and see the two planes in gazebo. I noticed there is plugins in the x340_base.xacro file, do you know what other files i need to incorporate?

mago3421, 2017-05-09:
@dennisss Both error logs are empty. Do you use any particular options in Dronekit's connect call?

dennisss, 2017-05-10:
@paan2097 The important part in the sdf file is that the mavlink plugin exists. If it works with stock px4 single drone simulation, then it should work here

@mago3421 I've never used Dronekit, but it should work with 127.0.0.1:14550 and 127.0.0.1:14560 etc.

Some other notes to try for both of you:
Make sure that the file `build_firmware/src/firmware/posix/px4` exists indicating that the firmware built completely

In the rcS file, the line that starts the simulator should be templated like this:
`simulator start -s -u _SIMPORT_`

The line that starts the mavlink interface should look like this:
`mavlink start -u _MAINPORT_ -r 4000000 -o _MAINPORTO_ -m onboard`, I think my original sample file had `-m magic`, although I've realized that this may break things depending on your environment.

paan2097, 2017-05-10:
@dennisss AH I switched over all the mavlink lines in the rcS file but didn't have the simulator start -s -u _SIMPORT_ line copied over, thanks. I'll be testing the multi plane SITL for awhile and will let you know of any issues. Thanks a bunch!

paan2097, 2017-05-12:
@dennisss Hey so I finally was able to get the planes to takeoff but they immediately crash, the err.log file is empty but here is an out.log file, the only difference between the two vehicles is that one had a positive tone_alarm.

data path: /home/pat/tansa/scripts/../lib/Firmware
commands file: rcS
creating new parameters file
creating new dataman file

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

ERROR [param] importing from 'rootfs/eeprom/parameters' failed (-1)
Command 'param' failed, returned 1
  MAV_TYPE: curr: 2 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 3033
18446744073709551608 WARNING: setRealtimeSched failed (not run as root?)
INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11797680 bytes
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1376264
  CAL_ACC0_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_ACC0_XSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC0_YOFF: curr: 0.0000 -> new: -0.0100
  CAL_ACC0_YSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC0_ZOFF: curr: 0.0000 -> new: 0.0100
  CAL_ACC0_ZSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC1_ID: curr: 0 -> new: 1310728
  CAL_ACC1_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_GYRO0_ID: curr: 0 -> new: 2293768
  CAL_GYRO0_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_MAG0_ID: curr: 0 -> new: 196616
  CAL_MAG0_XOFF: curr: 0.0000 -> new: 0.0100
  COM_RC_IN_MODE: curr: 0 -> new: 1
  MIS_LTRMIN_ALT: curr: 1.2000 -> new: 30.0000
  MIS_TAKEOFF_ALT: curr: 2.5000 -> new: 30.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 15.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_MAG_TYPE: curr: 0 -> new: 1
  FW_LND_ANG: curr: 5.0000 -> new: 8.0000
  FW_R_TC: curr: 0.4000 -> new: 0.7000
  FW_RR_FF: curr: 0.5000 -> new: 0.2000
  FW_RR_I: curr: 0.0100 -> new: 0.0200
  FW_RR_P: curr: 0.0500 -> new: 0.2200
  FW_P_TC: curr: 0.4000 -> new: 0.5000
  FW_PR_FF: curr: 0.5000 -> new: 0.4000
  FW_PR_I: curr: 0.0200 -> new: 0.0500
  FW_PR_P: curr: 0.0800 -> new: 0.0500
  FW_W_EN: curr: 0 -> new: 1
  RWTO_TKOFF: curr: 0 -> new: 1
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Waiting for initial data on UDP port 14571. Please start the flight simulator to proceed..
INFO  [simulator] Got initial simulation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
Sleeping for 1 s; (1000000 us).
.
.
INFO  [mavlink] mode: Magic, data rate: 2000000 B/s on udp port 14565 remote port 14560
INFO  [tone_alarm] startup
INFO  [mavlink] mode: Onboard, data rate: 2000000 B/s on udp port 14561 remote port 14566
WARN  [mavlink] bind failed: Address already in use
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: rootfs/fs/microsd/log
INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2017-05-12/16_10_19.ulg
INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 17, OBS buf: 16)
INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
INFO  [commander] home: 47.3977453, 8.5455957, 488.18
INFO  [tone_alarm] home_set
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
WARN  [navigator] offboard mission updated: dataman_id=1, count=0, current_seq=0
INFO  [tone_alarm] neutral
WARN  [navigator] offboard mission updated: dataman_id=0, count=3, current_seq=1
INFO  [tone_alarm] neutral
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977454, 8.5455958, 488.03
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977454, 8.5455958, 488.03
INFO  [tone_alarm] arming
INFO  [commander] Takeoff detected
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
WARN  [commander_tests] Failsafe enabled: no global position
INFO  [tone_alarm] negative
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] fast_bat

Exiting...
Shutting down
Restoring terminal

mzahana, 2017-05-13:
Thanks @anuppari and @dennisss for the suggestions and workarounds.

I am just thinking that it would be more useful to centralize the solution that works seamlessly with latest master onto PX4 repo. Is there a PR for this, with a clean documentation. Thanks.

mago3421, 2017-05-15:
@dennisss, what IP address and UDP port number can I use to connect to each instance using QGroundControl? I would like to monitor the planes and see why they are crashing.

mzahana, 2017-05-15:
@mago3421 could you check #7225 ?

mago3421, 2017-05-15:
Hi @mzahana, I don't see anything on #7225. Did you want me to post there?

mzahana, 2017-05-15:
Sorry, I meant this PR
https://github.com/PX4/Firmware/pull/7235#issuecomment-301314991

closed 2018-02-21 by PX4BuildBot