PX4/PX4-Autopilot #6415: ESC can't start

sunshine-yan, 2017-01-22:
Hi,
ESC can't start. I have tried to replace the "snapdragm_pwm_out start -d /dev/tty-2" with "uart_esc start -D /dev/tty-2". However,ESC still can't start. Is snapdragm_pwm_out about the ESC starting? 
The situation is as follows：
[image]

LorenzMeier, 2017-01-22:
@jywilson @bharath374 @mcharleb Advice?

sunshine-yan, 2017-01-22:
@LorenzMeier This is the ESC:
[image]

mhkabir, 2017-04-03:
Closing in favour of https://github.com/PX4/Firmware/issues/6962

closed 2017-04-03 by mhkabir