PX4/PX4-Autopilot #6422: Intel Aero: Add MAVLink crash dump via serial

LorenzMeier, 2017-01-23:
@zehortigoza We should add a MAVLink crash dump via serial. I'll merge https://github.com/PX4/Firmware/pull/6421 now but it would be good to be able to know what's going on for each hard fault.

LorenzMeier, 2017-01-28:
@bkueng @davids5 We need in general crash dumps in ulog so it would be good if you could connect in the next week and figure out how we dump the crash data to ulog. We probably should also log threads which are close to running out of stack and implement a proper warning about them.

The approach for crash dumps I have in mind is roughly like this:

  * After the reset, let the ulog app check if the crash flag is set
  * If it is set, find the last ulog file that was open (create one if none could be found)
  * Write a crash marker / fill so that the parser realises there is a new section (in case the file is corrupt) and then write the crash dump
  * Properly close the file and fsync it

We should tune this so this happens within ~200 ms, such that the crash dump is appended to the flight log before the vehicle hits the ground. This will maximise chances that we have a complete and readable log file. It probably also makes sense to start a new log so that we can see what happens (how it crashes and where).

TSC21, 2018-02-21:
@avinash-palleti are we able to get something done regarding this?

closed 2019-02-13 by stale[bot]