PX4/PX4-Autopilot #6423: pixracer stops responding after mission upload

dagar, 2017-01-23:
Pixracer running current PX4 stable FW with EKF2. 

    It is very easy to repeat, I can be in any mode and all that I have to do is reupload a
    mission that either has too steep of a glideslope or one where the land waypoint is
    higher than the previous waypoint. It is fairly random when it occurs, sometimes it 
    happens on the second upload sometimes on the sixth or seventh but it is repeatable.

Occurs on multiple pixracers.

original post - http://discuss.px4.io/t/pixracer-board-lockup-in-flight/2232/19

dagar, 2017-01-23:
I'll try and reproduce this locally.

Kjkinney, 2017-01-23:
We have reproduced it on the ground in a custom version of 1.5.1 and version 1.5.4 in both qgroundcontrol 3.0.2 and 3.1.0. This error has been reproduced across 3 separate aircraft all using pixracer.

dagar, 2017-01-24:
Would you mind trying the latest PX4 master as well? https://s3.amazonaws.com/px4-travis/Firmware/master/px4fmu-v4-default.px4

If it happens and auto resets check the sdcard for a hard fault log.

Kjkinney, 2017-01-24:
It won't trigger on px4 master. Switched to the current 1.5.4 and it happened right away.

LorenzMeier, 2017-01-25:
@Kjkinney Can you please re-test this with the latest stable? I have increased the navigator stack size, which is the likely cause. I could not reproduce locally.

Kjkinney, 2017-01-25:
Just tested with the latest stable and I could not reproduce it. I am getting a lot of operation timeout errors if the plane is armed however.

Kjkinney, 2017-01-25:
@LorenzMeier @dagar Tested version 1.5.1, 1.5.2, 1.5.3, 1.5.4 and all cause a lock-up roughly 1/4 of the time when a mission is uploaded with an invalid landing. We know this because we get the bad tones related to mission failure and we see the "WARN [navigator] Last waypoint to close to landing waypoint" or "Landing waypoint above last nav waypoint" in the console just prior to lock-up.  The lock-up can occur both when Armed and Disarmed on the ground.  Note: if we are inside with a poor GPS we do not get any tones when we upload a mission, valid or invalid. We cannot cause a lock-up in this condition is. Is this because the mission checker is not running yet?.  Maybe that is why you are unable to recreate it?

If we get a good tones and no errors after mission upload then everything seems fine.

Some good news is we cannot re-create the lock-up in 1.5.5 after > 50 tries on the ground.  Does any of the above information help pinpoint the true cause?  We'd feel more confortable knowing others can recreate this and that whatever changed in 1.5.5 really fixed the root cause.

LorenzMeier, 2017-01-25:
I will try to repro and close this when I could validate your results. Cause and fix are completely logical though, so I'm rather confident we nailed it.

dagar, 2017-01-26:
@Kjkinney this is definitely fixed by increasing the navigator stack. I was able to reproduce after 5 tries in 1.5.4, but not at all in 1.5.5

Navigator 1.5.4 stack 1300 bytes
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
 264 navigator                    1469  1.716  **1296/ 1296** 105 (105)  w:sem 

Navigator 1.5.5 stack 1500 bytes
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
 270 navigator                    6386  1.717  **1404/ 1496** 105 (105)  w:sem

closed 2017-01-26 by dagar