PX4/PX4-Autopilot #6426: Manual Stick control minimal lowpass filter

MaEtUgR, 2017-01-24:
When for example letting the spring loaded roll,pitch stick fall back to the center position, high frequency setpoint oscillations can occur.

These Setpoint fluctuations directly get to the controller and can drive the system unstable. I think there should be a minimal low pass filter on any user input. It should not influence the pilots intended input at all but filter all fast disturbances.

In my software I use 5Hz cutoff at the moment (see https://github.com/MaEtUgR/FlyBed/blob/master/main.cpp#L52) but this value can be adapted to personal preference and the result can easily be examined when looking at a low latency live plot of the filtered stick input at your hands.

MaEtUgR, 2017-01-25:
I started to work on this now on this branch: https://github.com/PX4/Firmware/tree/pr-rc-filtering .

dagar, 2017-01-25:
Unless you want this immediately I could add a filter as part of the stick_mapper (https://github.com/PX4/Firmware/issues/2822) I've been slowly picking away at. https://github.com/PX4/Firmware/compare/master...dagar:stick_mapper?expand=1

The idea is to centralize setpoint generation from stick input and get all the manual_control_setpoint usage out of the controllers. I was going to use BlockLowPass.

MaEtUgR, 2017-01-26:
I just started using `math::LowPassFilter2p` but your idea sounds like a good solution I appreciate.
What is your timeframe?
Maybe we could even cooperate to push further?

dagar, 2017-01-26:
Hopefully I'll find some time finish off the basic version this weekend.
The core stick mapper works for manual, acro, rattitude, stab, but I've mainly tested FW.
On the MC side I need to go through and make sure mc_pos_ctrl stops publishing during manual/stab.

The hardest part right now is figuring out how to handle VTOL, including switching between params that define input limits (MC <-> FW).

There's a lot more we can do here later. This is just the "basic" version that at least gets manual_control_setpoints out of the attitude controllers.

closed 2018-02-21 by PX4BuildBot

MaEtUgR, 2018-02-21:
There's filtering since #6440