PX4/PX4-Autopilot #6430: PX4FMU: Recongnized as COM port in Win 10

tubeme, 2017-01-24:
Hello,

I have the following problem. When I connect the Pixhawk to Win 10 the device is recognized as plain COM port. So I could not start the flashing.

I know that there are similar problems when connecting Pixhawk to USB hub. But in my case I connect it directly. Or could it be an internal hub that is messing the driver?

Any ideas?

lovettchris, 2017-01-25:
Assuming you have the PX4 driver installed, what do you see from the [system console](https://dev.px4.io/advanced-system-console.html)? 

You can get this if the PX4 is not booting properly, and the system console will show you why.  For example I can get this when I disable ekf2 and enable LPE but forget to change SYS_MC_EST_GROUP to 1.  Or vice versa.  When this happens the rcS boot script doesn't complete, so it doesn't launch mavlink and the usb driver that shows up as PX4.

tubeme, 2017-01-25:
The problem is that the device is booting properly and it is recognized as a com5 port. When I try to manually force the FMU driver the windows crashes.

The problem is that on my laptop everything is perfectly OK. I'm able to flash the board, it connects as PX4FMU etc.

tubeme, 2017-01-31:
Solved! Windows drivers problem... Thanks

closed 2017-01-31 by tubeme